ArduPilot/ardupilot_wiki #1553: Document Mavlink2 signing procedure

rmackay9, 2019-01-04:
We should probably add a section on how users can setup packet signing either in the [Developer MAVLink Interface section of the wiki](http://ardupilot.org/dev/docs/mavlink-commands.html) or better yet in the [user's telemetry section](http://ardupilot.org/copter/docs/common-telemetry-landingpage.html).

hamishwillee, 2019-01-05:
Generic information about signing should be linked from https://mavlink.io/en/guide/message_signing.html 

This draws heavily on the [proposal](https://docs.google.com/document/d/1ETle6qQRcaNWAmpG2wz0oOpFKSF_bcTmYMQvtTGI8ns/edit?usp=sharing) and has links to signing using the C library and Pymavlink.

The routing section briefly mentions impact of signing too. 

What is missing is the actual mechanism adopted by each autopilot for setting and storing the key, and which Ground stations support this mechanism. I believe for ArduPilot the code accepts **any** packet on the default USB port, so you can just connect and send the message to set the key from the GCS. I HOPE that the autopilot prevents automatic routing of the message to other non-private interfaces.

I would like to add a little "specific" information about what the flight stacks do, or at least a link to this new doc.

rmackay9, 2019-01-07:
discussion on this topic with Michael Oborne: https://discuss.ardupilot.org/t/enabling-mavlink-2-0-and-packet-signing/25656

rmackay9, 2020-02-20:
This came up again today that we don't have a page explaining how to set this up.

FYI @tridge, @Hwurzburg

auturgy, 2020-02-20:
https://ardupilot.github.io/MAVProxy/html/modules/signing.html

Regards,

James

Hwurzburg, 2020-05-12:
#2746   will add a link in telemetry today also

closed 2020-05-12 by Hwurzburg