ArduPilot/ardupilot_wiki #1560: Copter: move quick wiring diagrams to ardupilot hardware pages

rmackay9, 2019-01-07:
The [Copter's Assembly section](http://ardupilot.org/copter/docs/assembly-instructions.html) includes a few "quick wiring" pages for a very limited number of autopilots.

These should be moved to the specific flight controller pages under [AutoPilot Hardware Options](http://ardupilot.org/copter/docs/common-autopilots.html).

rmackay9, 2019-01-07:
After this is done I think we should promote the contents of the "Assembly Instructions" area (i.e. Mounting the Flight Controller, ESCs and Motors, GPS+Compass, Vibration Damping, Magnetic interference) up one level so it's under "First Time Setup'

Hwurzburg, 2019-11-01:
personally I am in favor of the hierarchical approach....I agree with the specific wiring diagram comment, but would like to keep a base wiring and required component subsytem assembly as a grouped topic....long lists are hard to select from...I am working to group things in the long TOC lists into logical , navigable groups...

Hwurzburg, 2020-01-15:
think this is in better shape now...

closed 2020-01-15 by Hwurzburg