ArduPilot/ardupilot_wiki #1565: Pozyx for Non-GPS Navigation

TOD9, 2019-01-08:
[Pozyx for Non-GPS Navigation](http://ardupilot.org/copter/docs/common-pozyx.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

TOD9, 2019-01-08:
the pozyx sometimes goes crazy, twice has gone full speed against the ceiling, I do not understand why, I have had problems with the compass that sometimes goes crazy and I would understand that I could go to the wrong side but not up. I have beacons configured on the north axis inside the square and with the jaw at -40 °, I do not know if this affects, I would like to know if anyone has had this problem and how to solve it.

rmackay9, 2019-01-09:
This list is for to-do items to add to the Wiki.  For support questions it would be better to raise this in the [Copter-3.6 forums](https://discuss.ardupilot.org/c/arducopter/copter-36) and include a dataflash log if possible.

closed 2019-01-09 by rmackay9