ArduPilot/ardupilot_wiki #1576: Telemetry-XBee page missing information

apache405, 2019-01-15:
[Telemetry-XBee](http://ardupilot.org/copter/docs/common-telemetry-xbee.html) 

Based on discussions in https://www.facebook.com/groups/thecubeautopilot/permalink/1953780218037924/, it looks like the wiki page for setting up Xbee radios is either out of date or missing information.

In order to get the link to work on my workbench with a [SparkFun XBee Explorer Regulated](https://www.sparkfun.com/products/11373), I had to remove the level shifters from the board to fix a signal level issue.

This image shows what the TX line from the Cube I was using looked with with the level shifter installed.  [image] 

This is what the Cube's TX line looked like after I removed the level shifters.
[image]

Here is documentation for the modification to the Sparkfun board.
[image]

The other item I needed to set was the Telem_Delay in the parameters.  I tested 30 and 20, but I suspect other values will work as well.

Otherwise, the directions in the wiki are ok, although they could use a reformating pass.

auturgy, 2019-01-15:
Could you do the update Jeff?

apache405, 2019-01-15:
@auturgy I will try to get it to this week.

auturgy, 2019-01-15:
Great, thanks!

Regards,

James

Hwurzburg, 2020-03-16:
I am not sure why removing the 3.3V to 5V bi-directional level shifters would be required with a flight controller UART port...should work with either 3.3 or 5V signalling as is..unless the CUBE has something special about its UART pins...I would like an explanation of why this impacts operation .....if CUBE needs 3.3V signalling then it needs to noted a LOT of places and would have shown up by now I would imagine...and in any case, board mod is not required since 3.3V level DOUT and DIN are already pinned out...

apache405, 2020-05-04:
@Hwurzburg the cube (and any STM32F427 based FC) is tolerant of 5V on the serial ports, but is a 3V3 system.  I'm not sure why the level shifters on the SFE board were misbehaving as the task I was working on did not cover root causing the problem.  If you look at the scope plot with the shifters installed you can see the waveform is not getting low enough to trigger V_IL conditions (0.3*VDD is a common specification for V_IL).