ArduPilot/ardupilot_wiki #1587: I added to the wiki of install ardupilot in linux sitl

josheeg, 2019-01-21:
I added to the wiki of install ardupilot in linux sitl

auturgy, 2019-01-21:
Hi,
I’m going to close this as you haven’t identified an issue.
I left a comment on one if your pr’s. If you can explain the problem you’re having we might be able to help more. Thanks for contributing though!

closed 2019-01-21 by auturgy

josheeg, 2019-01-22:
I have a fresh install of ubuntu linux 18.04 I go through the steps in the tutorial and added things as a reaction to what it threw errors about or comments warnings. I may have read into some things a bit too far and thought I needed them. I wanted to have the documentation good so it could be fallowed and some user might simulate a plane or quadplane and use jsbsim in mavproxy. Also be able to take off something other than a rascal. I think it is the only jsbsim plane that seems to takeoff but the notes say on flightgear is you don't have to start the rascal or up the throttle