ArduPilot/ardupilot_wiki #1599: Deprecated Documentation

stephendade, 2019-01-23:
Is it worth removing deprecated documentation? Or making it more obvious that it's deprecated?

I ask this after looking at the "Building the Code" section of the developer documentation. There's still many pages there that have a small "Deprecated" label in the left-hand navigation bar ... and that's it (example http://ardupilot.org/dev/docs/building-for-navio-on-rpi2.html). Not really obvious.

In order to reduce the chance of new developers/users mistakenly using the old documentation, do we want to:
a) Remove the deprecated pages (the cleanest way, but maybe people want the old documentation for reference?)
b) Adding a "deprecated" Section to the documentation navigation bar and putting all the pages there
c) Just adding a big "deprecated" warming to the top of each page?

khancyr, 2019-01-23:
Even if those doc are deprecated there are still people using then... A bunch could be removed thus...
I would vote for option C for now and try to make some triage on which on could be removed

rmackay9, 2019-01-23:
I also vote for (c)

stephendade, 2019-01-23:
>(b) is already done I think. http://ardupilot.org/copter/docs/common-archived-topics.html

Ahh, I did not see that before. In that case, probably best to extend this across to the dev documentation

hamishwillee, 2019-01-30:
Using Appendix for deprecation was a good temporary fix. A better long term solution is to provide versioned builds - that would allow us to dump all the cruft and just have relevant/recommended stuff in the head revision.
This is perfectly do-able, but requires a degree of automation of the toolchain.

Hwurzburg, 2019-11-01:
versioning is beyond the scope of this issue and is always being discussed, so I am closing this specific wiki issue...we are trying to stay current on Archived topics...

closed 2019-11-01 by Hwurzburg