ArduPilot/ardupilot_wiki #1601: C.H.U.C.K Autopilot (for Parrot Disco)

MEXICO-PARIS, 2019-01-24:
[C.H.U.C.K Autopilot (for Parrot Disco)](http://ardupilot.org/plane/docs/common-CHUCK-overview.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

proficnc, 2019-01-24:
What is the actual issue?

rmackay9, 2019-01-24:
perhaps a mistake so closing..

closed 2019-01-24 by rmackay9

auturgy, 2019-01-27:
> je recherche le firmware AG parrot disco

http://firmware.ardupilot.org/Plane/stable/disco/

Regards,

James

MEXICO-PARIS, 2019-01-27:
bonjour Jame

le firmware et il pour la sequoia du parrot disco pro

auturgy, 2019-01-27:
Bonjour,

Je ne sais pas si le Pro avec séquoia est le même.
Pierre parle français. J'espère qu'il peut aider davantage. @khancyr

Regards,

James

MEXICO-PARIS, 2019-01-27:
ok merci James

khancyr, 2019-01-28:
Bonjour,

Normalement, le disco pro est sur le firmware de Parrot pas ArduPilot. Malheuresement, nous n'avons pas d'API pour le control de la camera sequoia. Le mieux est de vous tourner vers le service client de Parrot.