ArduPilot/ardupilot_wiki #1627: Setting up the Build Environment (MacOSX)

priseborough, 2019-02-05:
[Setting up the Build Environment (MacOSX)](http://ardupilot.org/dev/docs/building-setup-mac.html) 

Warning generated when running 'brew install gawk' step, see below:

> 
==> Installing dependencies for gawk: gettext, gmp, mpfr and readline
==> Installing gawk dependency: gettext
==> Downloading https://homebrew.bintray.com/bottles/gettext-0.19.8.1.mojave.bot
######################################################################## 100.0%
==> Pouring gettext-0.19.8.1.mojave.bottle.tar.gz
==> Caveats
gettext is keg-only, which means it was not symlinked into /usr/local,
because macOS provides the BSD gettext library & some software gets confused if both are in the library path.

If you need to have gettext first in your PATH run:
  echo 'export PATH="/usr/local/opt/gettext/bin:$PATH"' >> ~/.bash_profile

For compilers to find gettext you may need to set:
  export LDFLAGS="-L/usr/local/opt/gettext/lib"
  export CPPFLAGS="-I/usr/local/opt/gettext/include"

==> Summary
🍺  /usr/local/Cellar/gettext/0.19.8.1: 1,935 files, 16.9MB
==> Installing gawk dependency: gmp
==> Downloading https://homebrew.bintray.com/bottles/gmp-6.1.2_2.mojave.bottle.t
######################################################################## 100.0%
==> Pouring gmp-6.1.2_2.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/gmp/6.1.2_2: 18 files, 3.1MB
==> Installing gawk dependency: mpfr
==> Downloading https://homebrew.bintray.com/bottles/mpfr-4.0.1.mojave.bottle.ta
######################################################################## 100.0%
==> Pouring mpfr-4.0.1.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/mpfr/4.0.1: 28 files, 4.6MB
==> Installing gawk dependency: readline
==> Downloading https://homebrew.bintray.com/bottles/readline-8.0.0.mojave.bottl
######################################################################## 100.0%
==> Pouring readline-8.0.0.mojave.bottle.tar.gz
==> Caveats
readline is keg-only, which means it was not symlinked into /usr/local,
because macOS provides the BSD libedit library, which shadows libreadline.
In order to prevent conflicts when programs look for libreadline we are
defaulting this GNU Readline installation to keg-only.

For compilers to find readline you may need to set:
  export LDFLAGS="-L/usr/local/opt/readline/lib"
  export CPPFLAGS="-I/usr/local/opt/readline/include"

==> Summary
🍺  /usr/local/Cellar/readline/8.0.0: 48 files, 1.5MB
==> Installing gawk
==> Downloading https://homebrew.bintray.com/bottles/gawk-4.2.1_1.mojave.bottle.
######################################################################## 100.0%
==> Pouring gawk-4.2.1_1.mojave.bottle.1.tar.gz
🍺  /usr/local/Cellar/gawk/4.2.1_1: 87 files, 4.7MB
==> Caveats
==> gettext
gettext is keg-only, which means it was not symlinked into /usr/local,
because macOS provides the BSD gettext library & some software gets confused if both are in the library path.

If you need to have gettext first in your PATH run:
  echo 'export PATH="/usr/local/opt/gettext/bin:$PATH"' >> ~/.bash_profile

For compilers to find gettext you may need to set:
  export LDFLAGS="-L/usr/local/opt/gettext/lib"
  export CPPFLAGS="-I/usr/local/opt/gettext/include"

==> readline
readline is keg-only, which means it was not symlinked into /usr/local,
because macOS provides the BSD libedit library, which shadows libreadline.
In order to prevent conflicts when programs look for libreadline we are
defaulting this GNU Readline installation to keg-only.

For compilers to find readline you may need to set:
  export LDFLAGS="-L/usr/local/opt/readline/lib"
  export CPPFLAGS="-I/usr/local/opt/readline/include"

> 

echo $PATH gives:

> 
/usr/local/bin:/Users/paul/bin:/Users/paul/Library/Python/2.7/bin:/usr/local/bin:/Users/paul/bin:/Users/paul/Library/Python/2.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin

> 

Running step 'sudo easy_install pip' gives the following error:

> 
Traceback (most recent call last):
  File "/usr/bin/easy_install-2.7", line 11, in <module>
    load_entry_point('setuptools==18.5', 'console_scripts', 'easy_install')()
  File "/Library/Python/2.7/site-packages/pkg_resources/__init__.py", line 476, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/Library/Python/2.7/site-packages/pkg_resources/__init__.py", line 2700, in load_entry_point
    return ep.load()
  File "/Library/Python/2.7/site-packages/pkg_resources/__init__.py", line 2318, in load
    return self.resolve()
  File "/Library/Python/2.7/site-packages/pkg_resources/__init__.py", line 2324, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/setuptools/command/easy_install.py", line 46, in <module>
    from setuptools.archive_util import unpack_archive
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/setuptools/archive_util.py", line 15, in <module>
    from pkg_resources import ensure_directory, ContextualZipFile
ImportError: cannot import name ContextualZipFile

>

priseborough, 2019-02-05:
SystemSW details:

`System Version:	macOS 10.14.3 (18D42)
  Kernel Version:	Darwin 18.2.0
  Boot Volume:	Macintosh HD
  Boot Mode:	Normal
  Computer Name:	Paul’s MacBook Pro
  Username:	Paul Riseborough (paul)
  Secure Virtual Memory:	Enabled
  System Integrity Protection:	Enabled
  Time since boot:	32 minutes
`

peterbarker, 2019-02-05:
... some small amount of search seemed to indicate `sudo -H python -m ensurepip` is worth a shot.