ArduPilot/ardupilot_wiki #1631: Setting up the Build Environment (Linux/Ubuntu)

priseborough, 2019-02-09:
[Setting up the Build Environment (Linux/Ubuntu)](http://ardupilot.org/dev/docs/building-setup-linux.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Updating install of build environment  prerequisites on Ubintu18.04LTS running on Parallels 14.11 generates the following error.

parallels@parallels-Parallels-Virtual-Platform:~/SourceCode/ardupilot$ Tools/scripts/install-prereqs-ubuntu.sh -y
---------- Tools/scripts/install-prereqs-ubuntu.sh start ----------
+ OPT=/opt
+ BASE_PKGS='build-essential ccache g++ gawk git make wget'
+ PYTHON_PKGS='future lxml pymavlink MAVProxy'
+ PX4_PKGS='python-argparse openocd flex bison libncurses5-dev           autoconf texinfo libftdi-dev zlib1g-dev           zip genromfs python-empy cmake cmake-data'
+ ARM_LINUX_PKGS='g++-arm-linux-gnueabihf pkg-config-arm-linux-gnueabihf'
+ SITL_PKGS='libtool libxml2-dev libxslt1-dev python-dev python-pip python-setuptools python-matplotlib python-serial python-scipy python-opencv python-numpy python-pyparsing xterm'
+ ASSUME_YES=false
+ QUIET=false
++ uname -m
+ MACHINE_TYPE=x86_64
+ '[' x86_64 == x86_64 ']'
+ PX4_PKGS+=' libc6-i386'
+ ARM_ROOT=gcc-arm-none-eabi-6-2017-q2-update
+ ARM_TARBALL=gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2
+ ARM_TARBALL_URL=http://firmware.ardupilot.org/Tools/STM32-tools/gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2
+ ARDUPILOT_TOOLS=Tools/autotest
+ OPTIND=1
+ getopts y opt
+ case "$opt" in
+ ASSUME_YES=true
+ getopts y opt
+ APT_GET='sudo apt-get'
+ true
+ APT_GET='sudo apt-get --assume-yes'
+ false
+ dpkg-query -l lsb-release
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version       Architecture  Description
+++-================-=============-=============-======================================
ii  lsb-release      9.20170808ubu all           Linux Standard Base version reporting 
+ read -r UBUNTU_CODENAME
++ lsb_release -c -s
+ '[' bionic = precise ']'
+ '[' bionic = trusty ']'
+ sudo usermod -a -G dialout parallels
+ dpkg-query -l modemmanager
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version       Architecture  Description
+++-================-=============-=============-======================================
rc  modemmanager     1.6.8-2ubuntu amd64         D-Bus service for managing modems
+ sudo apt-get --assume-yes remove modemmanager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'modemmanager' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 399 not upgraded.
+ sudo apt-get --assume-yes update
Hit:1 http://us.archive.ubuntu.com/ubuntu bionic InRelease
Hit:2 http://us.archive.ubuntu.com/ubuntu bionic-updates InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu bionic-backports InRelease                     
Hit:4 http://security.ubuntu.com/ubuntu bionic-security InRelease                        
Reading package lists... Done
+ apt-cache search python-wxgtk3.0
+ grep wx
python-wxgtk3.0 - Python interface to the wxWidgets Cross-platform C++ GUI toolkit
python-wxgtk3.0-dev - Development files for wxPython
+ SITL_PKGS+=' python-wxgtk3.0 libtool-bin'
++ apt-cache search -n '^realpath$'
+ RP=
+ '[' -n '' ']'
+ sudo apt-get --assume-yes install build-essential ccache g++ gawk git make wget libtool libxml2-dev libxslt1-dev python-dev python-pip python-setuptools python-matplotlib python-serial python-scipy python-opencv python-numpy python-pyparsing xterm python-wxgtk3.0 libtool-bin python-argparse openocd flex bison libncurses5-dev autoconf texinfo libftdi-dev zlib1g-dev zip genromfs python-empy cmake cmake-data libc6-i386 g++-arm-linux-gnueabihf pkg-config-arm-linux-gnueabihf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libpython2.7-stdlib' instead of 'python-argparse'
autoconf is already the newest version (2.69-11).
bison is already the newest version (2:3.0.4.dfsg-1build1).
build-essential is already the newest version (12.4ubuntu1).
ccache is already the newest version (3.4.1-1).
cmake is already the newest version (3.10.2-1ubuntu2).
cmake-data is already the newest version (3.10.2-1ubuntu2).
flex is already the newest version (2.6.4-6).
gawk is already the newest version (1:4.1.4+dfsg-1build1).
libc6-i386 is already the newest version (2.27-3ubuntu1).
libtool is already the newest version (2.4.6-2).
libtool-bin is already the newest version (2.4.6-2).
libxslt1-dev is already the newest version (1.1.29-5).
make is already the newest version (4.1-9.1ubuntu1).
python-dev is already the newest version (2.7.15~rc1-1).
python-numpy is already the newest version (1:1.13.3-2ubuntu1).
python-pyparsing is already the newest version (2.2.0+dfsg1-2).
python-serial is already the newest version (3.4-2).
python-setuptools is already the newest version (39.0.1-2).
zip is already the newest version (3.0-11build1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-0ubuntu2).
genromfs is already the newest version (0.5.2-2build3).
libftdi-dev is already the newest version (0.20-4build3).
openocd is already the newest version (0.10.0-4).
python-empy is already the newest version (3.3.2-1build1).
python-matplotlib is already the newest version (2.1.1-2ubuntu3).
python-scipy is already the newest version (0.19.1-2ubuntu1).
python-wxgtk3.0 is already the newest version (3.0.2.0+dfsg-7).
texinfo is already the newest version (6.5.0.dfsg.1-2).
xterm is already the newest version (330-1ubuntu2).
g++ is already the newest version (4:7.3.0-3ubuntu2.1).
g++-arm-linux-gnueabihf is already the newest version (4:7.3.0-3ubuntu2.1).
libncurses5-dev is already the newest version (6.1-1ubuntu1.18.04).
libxml2-dev is already the newest version (2.9.4+dfsg1-6.1ubuntu1.2).
wget is already the newest version (1.19.4-1ubuntu2.1).
pkg-config-arm-linux-gnueabihf is already the newest version (4:7.3.0-3ubuntu2.1).
python-opencv is already the newest version (3.2.0+dfsg-4ubuntu0.1).
python-pip is already the newest version (9.0.1-2.3~ubuntu1).
The following additional packages will be installed:
  libpython2.7 libpython2.7-dev libpython2.7-minimal python2.7 python2.7-dev
  python2.7-minimal
Suggested packages:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email gitweb git-cvs
  git-mediawiki git-svn python2.7-doc binfmt-support
The following packages will be upgraded:
  git libpython2.7 libpython2.7-dev libpython2.7-minimal libpython2.7-stdlib python2.7
  python2.7-dev python2.7-minimal
8 upgraded, 0 newly installed, 0 to remove and 391 not upgraded.
Need to get 0 B/37.3 MB of archives.
After this operation, 98.3 kB of additional disk space will be used.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libwebkit2gtk-4.0-37:amd64' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

peterbarker, 2019-02-09:
On Fri, 8 Feb 2019, Paul Riseborough wrote:

>     After this operation, 98.3 kB of additional disk space will be used.
>     dpkg: unrecoverable fatal error, aborting:
>     files list file for package 'libwebkit2gtk-4.0-37:amd64' is missing final newline
>     E: Sub-process /usr/bin/dpkg returned an error code (2)

OK, that's a new one on me.  Is it repeatable?

Hwurzburg, 2020-01-15:
stale issue....@priseborough  Paul please reopen if you have more info and its repeatable...

closed 2020-01-15 by Hwurzburg