ArduPilot/ardupilot_wiki #1657: SiK Radio — Advanced Configuration

AndKe, 2019-02-27:
[SiK Radio — Advanced Configuration](http://ardupilot.org/copter/docs/common-3dr-radio-advanced-configuration-and-technical-information.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

the line:
"The radio firmware supports both the MAVLink 0.9 and the MAVLink 1.0 transmission formats."

MAVLink 2.0 deserves a mention as well.

brunoolivieri, 2019-06-06:
Hi, 

Thanks for the observation. Indeed I got updated information on forums [1], and I did the PR #1780

Could we close this issue?

[1] https://discuss.ardupilot.org/t/sik-radios-vs-mavlink-versions/43282/2?u=brunoolivieri

AndKe, 2019-06-06:
yes, thank you.

closed 2019-06-06 by AndKe