ArduPilot/ardupilot_wiki #1678: Altitude Hold Mode  -mentioned parameter INAV_TC_Z does not exist.

AndKe, 2019-03-15:
[Altitude Hold Mode](http://ardupilot.org/copter/docs/altholdmode.html) -

INAV_TC_Z  does not exist, please update with proper solution.

rmackay9, 2019-03-15:
@AndKe, the equivalent parameter in the EKF is the EK2_ALT_M_NSE.  It's mentioned on [this user focused EKF wiki page](http://ardupilot.org/copter/docs/common-apm-navigation-extended-kalman-filter-overview.html) but I'll update the AltHold page in the near-ish future.

brunoolivieri, 2019-06-11:
Hi @rmackay9	

I understood that the parameter INAV_TC_Z is for EKF1 and EK2_ALT_M_NSE for EKF2. Am I correct?

If so, could I update the information?

rmackay9, 2019-06-12:
@brunoolivieri, yes, that's right.  I'm pretty sure the scaling is not the same but it is the equivalent..

brunoolivieri, 2019-06-12:
@rmackay9 Thanks! :-)

@AndKe I created the PR #1793 to fix that. Are you ok to close the PR?

AndKe, 2019-06-12:
Yes, thank you

closed 2019-06-12 by AndKe