ArduPilot/ardupilot_wiki #1681: Smart RTL Mode:  what fs too many points are requested ?

AndKe, 2019-03-18:
[Smart RTL Mode](http://ardupilot.org/copter/docs/smartrtl-mode.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Please describe risk/what will happen if user reserves memory for too many SRTL_POINTS , is it possible to crash/negatively affect other autopilot functions ?

Hwurzburg, 2019-10-10:
@rmackay9...given that 500 points is the maximum now, is there any chance on any flight controller of this? btw, I will update the page to include the maximum number and the new warnings on buffer space running low today...the one code PR that I forgot to update in the wiki....

rmackay9, 2019-10-10:
@AndKe,

SmartRTL internally forces a maximum of 500 points even if the user sets the number higher than this.  The buffers are allocated at startup and if it runs out of memory it handles it gracefully so i don't think there will be problems in flight.

closed 2019-10-10 by rmackay9