ArduPilot/ardupilot_wiki #1704: Combine training centers and school pages

rmackay9, 2019-04-09:
We apparently have separate training centers and school wiki pages.  we should combine these.

auturgy, 2019-04-09:
Originally they were intended for different things: flight/operations training centres, and code/development schools.
I think the distinction made sense, until we cross merged and messed it up.
Up to you though.

Regards,

James

brunoolivieri, 2019-06-02:
I can see the Training Centers in plane/copter wikis but Schools only in dev wiki. None of the others have those links. It is correct? Should be updated?

rmackay9, 2019-06-02:
@brunoolivieri, yes, they should all be consistent I think.

Hwurzburg, 2019-10-10:
It would seem that code schools and flight training should be separated... and both pages listed in all wikis...I will do that if you guys agree...

rmackay9, 2019-10-10:
I think we should just put them together on a single page with some way to see what each school offers.  Otherwise I think we will end up with duplicates as some schools offer both... and i think from the user's point of view they are all schools.

Hwurzburg, 2019-10-11:
PR #2066 combines

closed 2019-12-22 by Hwurzburg