ArduPilot/ardupilot_wiki #1708: Complete Parameter List: INS_PRODUCT_ID is undocumented

AndKe, 2019-04-12:
[Complete Parameter List](http://ardupilot.org/copter/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

subject says all.

rmackay9, 2019-04-12:
[Apparently it was an unused parameter](https://github.com/ArduPilot/ardupilot/pull/10099) so it's been removed along with it's documentation.  We have a known issue re parameter descriptions coming from master instead of the branch corresponding to the release.

I might close this because the versioning issue is already documented and I don't think we will add back parameter descriptions for this parameter to master.

brunoolivieri, 2019-06-02:
Hi! I believe that this issue could be closed due to PR 10099 (https://github.com/ArduPilot/ardupilot/pull/10099)
 
Regards,

closed 2019-06-02 by WickedShell