ArduPilot/ardupilot_wiki #1710: Companion Computers few page out of date

bys1123, 2019-04-16:
1. Disable bluetooth on RPi3 should use `dtoverlay=pi3-disable-bt` in /boot/config.txt instead of `dtoverlay=pi-disable-bt` descripted in wiki.
http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html#testing-the-connection
2. Look like Config page is work on 80 port with apweb, (8000 port is only for video?)
http://ardupilot.org/dev/docs/apsync-intro.html#simple-configuration-web-page

bys1123, 2019-04-16:
I saw Randy mentioned apweb on #1147

brunoolivieri, 2019-06-10:
Hi @bys1123 

I did the simple PR #1786 to fix the BT command and PR #1785 to insert item "2" in the Wiki.

Could this issue be closed?

rmackay9, 2019-06-11:
@brunoolivieri, thanks for all your efforts on the wiki, txs!  i'll close this now.

closed 2019-06-11 by rmackay9