ArduPilot/ardupilot_wiki #1713: Using an Airspeed Sensor

wagnera, 2019-04-17:
[Using an Airspeed Sensor](http://ardupilot.org/plane/docs/airspeed.html) - This should be updated that the ARSPD_PIN should be 4 for the Pixhawk 4. 11 or 15 is not correct.

brunoolivieri, 2019-06-11:
Hi @wagnera

In order to allow me to edit the wiki properly, please:

1) "ARSPD_PIN should be 4 for the Pixhawk 4" Are you talking about analog or digital sensor? I believe that information is just for the analog sensor.

2) Any other differences from standard Pixhawk that should be mentioned?

kd0aij, 2019-06-11:
By "PX4" that page is referring to the original PX4FMU and PX4IO 2-board set.
I don't see any mention of "Pixhawk 4" at all.

wagnera, 2019-06-11:
Correct for the analog airspeed sensor connected to ADC-6V6.  
Another note would be that when using the power management [board](http://www.holybro.com/manual/PM07-Quick-Start-Guide.pdf) the FMU PWM-OUT pins labeled 1-8 correspond to servos 9-16 in the parameters since M1-M8 are servos 1-8.

brunoolivieri, 2019-06-14:
> By "PX4" that page is referring to the original PX4FMU and PX4IO 2-board set.
> I don't see any mention of "Pixhawk 4" at all.

@kd0aij, sure, these are different boards. Do you see any problem to insert such detail about Pixhawk 4?

brunoolivieri, 2019-06-14:
> Correct for the analog airspeed sensor connected to ADC-6V6.
> Another note would be that when using the power management [board](http://www.holybro.com/manual/PM07-Quick-Start-Guide.pdf) the FMU PWM-OUT pins labeled 1-8 correspond to servos 9-16 in the parameters since M1-M8 are servos 1-8.

@wagnera, Thank you. 

However, such detail regarding power should be inserted in a whole new page for Pixhawk 4 as there is a page for the Pixhawk 1. Am I correct? 

Pixhawk 1: http://ardupilot.org/plane/docs/common-pixhawk-overview.html

wagnera, 2019-06-14:
Yes it should be on a whole new page like the one for the Pixhawk 1.

kd0aij, 2019-06-14:
I think a new page for Pixhawk 4 would be good.
"PX4/Pixhawk" on the existing page should probably be clarified to avoid confusion, though.

brunoolivieri, 2019-06-17:
Indeed, I do believe that a whole new page for Pixhawk v4 is necessary. 

My propose is:

1) Add airspeed information regarding the correct PINs in http://ardupilot.org/plane/docs/airspeed.html . (I did the PR #1803  )

2) Create the task to create the new page. (Issue #1804 )

If you agree, we could close this issue

brunoolivieri, 2019-06-24:
Actually, the repo already have information about the Pixhawk 4 linked in http://ardupilot.org/copter/docs/common-autopilots.html#closed-hardware under the name "Holybro Pixhawk 4" linked to https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/Pixhawk4/README.md

Hwurzburg, 2019-10-10:
Resolved with the above commit...closing

closed 2019-10-10 by Hwurzburg