ArduPilot/ardupilot_wiki #1719: Building for Bebop on Linux

malfonsoNeoris, 2019-04-23:
[Building for Bebop on Linux](http://ardupilot.org/dev/docs/building-for-bebop-on-linux.html) 
Hi.. i would like no know if there are updates regarding the bebop or bebop 2 ardupilot.

- [ ] Optical Flow. or at list stable indoor hover like in the original autopilot
- [ ] Camera streaming.. or a way to get frontal camera from python program

thanks

brunoolivieri, 2019-06-02:
Hi @malfonsoNeoris , I believe that should be a question to discuss in the forums (https://discuss.ardupilot.org). Could you try it?

I believe that this issue could be closed due to the fact that is not a wiki issue.

Regards,

rmackay9, 2019-06-02:
I think camera streaming is supported on the Disco but I'm less sure if the Bebop2 has the correct library.  In any case, as @brunoolivieri says, it's not a wiki issue so closing.

closed 2019-06-02 by rmackay9

malfonso0, 2019-06-03:
sorry for the misplaced question
i will ask in the forum