ArduPilot/ardupilot_wiki #1733: Copter SITL/MAVProxy Tutorial : Wrong instructions

Orbitography, 2019-05-03:
There are 2 lines supposed to start SITL : 

cd ~/ardupilot/ArduCopter
sim_vehicle.py --map --console

They send back an error because there is no sim_vehicle.py in ~/ardupilot/ArduCopter
There is a sim_vehicle.py in ~/ardupilot/Tools/autotest but trying in on this file send an error message : 

SIM_VEHICLE: Start
SIM_VEHICLE:
** Is (autotest) really your vehicle type?
Perhaps you could try -v ArduCopter|AntennaTracker|APMrover2|ArduSub|ArduPlane
You could also try changing directory to e.g. the ArduCopter subdirectory

SIM_VEHICLE: Killing tasks

What should be the right instructions in the Wiki ( http://ardupilot.org/dev/docs/copter-sitl-mavproxy-tutorial.html )

kd0aij, 2019-05-03:
I create a directory below the vehicle directory e.g. ArduCopter/sitl_test
and then run ../../Tools/autotest/sim_vehicle.py from there (as suggested by the usage info)
This results in an eeprom.bin and logs subdirectory (and other files) located in ArduCopter/sitl_test

brunoolivieri, 2019-06-02:
Hi, I believe that is solved.
I believe that this issue could be closed.
regards,

rmackay9, 2019-06-02:
I've slightly fixed the instructions so it says to run ../Tools/autotest/sim_vehicle.py (i.e. prepended the path to the correct directory).  Txs for the report!

closed 2019-06-02 by rmackay9