ArduPilot/ardupilot_wiki #1739: Choosing Which Build Environment on Windows (reasons for options please)

waynemcl, 2019-05-08:
[Setup the Build Environment on Windows](http://ardupilot.org/dev/docs/building-setup-windows.html) - 
Can a quick list of reasons (pros / cons) be given for choosing which build environment to use please?
Users interested compiling / contributing code may not know which of the options would suit 
 their needs / working environment best. e.g. I'm on Win10, but familiar with linux and arduino. This may prevent a few trial / error cycles working this out for us. 
Thanks devs!

brunoolivieri, 2019-06-02:
Hi @waynemcl , I believe that should be a question to discuss in the forums (https://discuss.ardupilot.org).

I believe that this issue could be closed.

Cheers.

rmackay9, 2019-06-02:
Re the original question, I don't know of particular advantages or disadvantages of each method.  I think everything works on both.  Personally I've moved to using cygwin but I don't think there's any particular advantage.

rmackay9, 2019-06-02:
I've updated the wiki to make it a bit more clear that both are supported and work equally well.

closed 2019-06-02 by rmackay9