ArduPilot/ardupilot_wiki #1742: Battery Failsafe

AndKe, 2019-05-11:
[Battery Failsafe](http://ardupilot.org/copter/docs/failsafe-battery.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Please describe behaviour if landing in RTL, or LAND mode and the battery failsafe triggers, will it switch back to RTL, climb again, etc ?
Same goes for other failsafe during LAND.

Naterater, 2019-06-02:
https://github.com/ArduPilot/ardupilot/issues/10320

The craft will continue to land.

I've made the change and proposed it here: https://github.com/ArduPilot/ardupilot_wiki/pull/1762

rmackay9, 2019-06-02:
@AndKe, think this issue is closed now?

brunoolivieri, 2019-07-09:
PR was merged. I do believe that can be closed.

closed 2019-07-09 by rmackay9