ArduPilot/ardupilot_wiki #1747: CUAV v5 Overview

StephBMG, 2019-05-15:
[CUAV v5 Overview](http://ardupilot.org/copter/docs/common-pixhackV5-overview.html) -
8 FMU PWM outputs (D-Shot capable)
I think it's wrong, there are only 6 FMU PWM outputs (D-Shot capable) as shown here https://github.com/ArduPilot/ardupilot/blob/7c2dc84d45f3bfad08db03312a5485627859a532/libraries/AP_HAL_ChibiOS/hwdef/CUAVv5/README.md

Naterater, 2019-06-02:
There are actually a total of 14 (8 main FMU + 6 aux "IOMCU"). It's technically correct.

StephBMG, 2019-06-02:
I've tested the CUAV V5, there is only 6 D-shot outputs not 8
you can see here 6 FMU AUX and 8 IO
[image]
https://github.com/cuav/hardware/blob/master/V5_Autopilot/V5_Ver2/BASE/V5_BASE_V2.1.pdf

brunoolivieri, 2019-06-02:
Hi @StephBMG , I believe that should be an issue on the main repository since that file does not belong to the wiki.

I believe that this issue could be closed.

regards,

rmackay9, 2019-06-02:
@mirkix has fixed this so closing, thanks!

closed 2019-06-02 by rmackay9