ArduPilot/ardupilot_wiki #1754: Firmware Website down

MpxFlyer, 2019-05-31:
I just wanted to update my cube. However I just got error messages like "no firmware availiable". I then wanted to download the FW manually via http://firmware.ardupilot.org 
However I cannot seem to access the website. Is it possible that it is down?

auturgy, 2019-05-31:
Yes. Firmware.ardupilot.org is currently down. It’s being looked at right now.

Regards,

James

peterbarker, 2019-05-31:
On Fri, 31 May 2019, MpxFlyer wrote:

> However I cannot seem to access the website. Is it possible that it is down?

Fixed!

Our bad.

MpxFlyer, 2019-05-31:
Awesome thanks for your quick reply. I just wanted to make sure it was not an error with my network or whatsoever. please keep us updated - looking forward to finally beeing able to try my pixhawk!

closed 2019-06-01 by OXINARF