ArduPilot/ardupilot_wiki #1756: Wiki PR cleanup status list

Hwurzburg, 2019-06-01:
| PR | Date|Status|Review needed by|
|:---:|:---:|:---:|:---:|
|#549| 10/16 |Close?||
|#694| 01/17| Review| Rmackay|
|#719| 07/17| Close?||
|#885 |07/17 |Close?||
|#892| 07/17| Decision| Rmackay|
|#962| 10/17 |Awaiting change||
|#970/#971/#991| 10/17 |Review |Oxinarf|
|#979||10/17| Awating change||
|#1032 |12/17 |Close?||

rmackay9, 2019-06-03:
Thanks very much for the efforts on this!

Hwurzburg, 2026-08-28:
Closing this as obsolete. 4.1 is five releases behind current, and the substantive items on this list have since been documented — NoopLoop and UPFlow both have their own pages, external AHRS is covered, DO_LAND_START and DO_REPOSITION appear in the mission planning docs, and so on.

Anything from the 4.1 cycle that genuinely never got written up will surface as its own issue rather than being found on a seven-year-old checklist. Later releases are tracked in #4274 (<4.8) and #7475 (4.8).

Feel free to reopen if you feel this has not been addressed.

closed 2026-08-28 by Hwurzburg