ArduPilot/ardupilot_wiki #1761: Wiki PR cleanup status list #2

vierfuffzig, 2019-06-01:
as discussed on wiki gitter. will be updated. trying my best to sort, adress and implement suggestions, thanks everyone pinged for looking into these:

https://github.com/ArduPilot/ardupilot_wiki/issues/1469 implemented by https://github.com/ArduPilot/ardupilot_wiki/pull/1757 @rmackay9 
https://github.com/ArduPilot/ardupilot_wiki/issues/1467 solved, close
#1461 implemented by https://github.com/ArduPilot/ardupilot_wiki/pull/1758 @rmackay9 
#1435 implented by https://github.com/ArduPilot/ardupilot_wiki/pull/1759 @khancyr 
#1433 implemented by https://github.com/ArduPilot/ardupilot_wiki/pull/1760 pretty specific, merge or close both ping @rmackay9

vierfuffzig, 2019-09-11:
updated, thanks everyone for your effort! 
can't close issues, got no permission, ping @Hwurzburg maybe...?

thanks, basti.

Hwurzburg, 2019-09-11:
will do

vierfuffzig, 2019-09-11:
thx henry!

vierfuffzig, 2019-09-11:
closing list with just the SITL diagram update left as WIP. thanks everyone!

closed 2019-09-11 by vierfuffzig