ArduPilot/ardupilot_wiki #1763: Complete Parameter List Versioning

Naterater, 2019-06-02:
This is a formal request for the Complete parameter list to be versioned by release. People are asking today where features and parameters are when they haven't yet made it to stable releases. The current complete parameter lists are auto generated from master, and I think it should be generated from the most recent stable release if there is only one version.

brunoolivieri, 2019-06-02:
Hi @Naterater , would not this be the case of a discussion in the forum?

I believe that this issue could be closed.

Cheers.

rmackay9, 2019-06-02:
I think I'll leave this one open.  It's a bit of a mix of flight code and wiki changes but it makes sense to track it here.

brunoolivieri, 2019-06-03:
ok

brunoolivieri, 2020-04-15:
The versioning for parameters is running. You can check it out: https://ardupilot.org/copter/docs/parameters.html  (it is a listbox of versions...)

I will close that, please feel free to reopen if necessary.

closed 2020-04-15 by brunoolivieri