ArduPilot/ardupilot_wiki #1788: firmware.ardupilot.org is down (servers are not available from June 8)

dzotov2, 2019-06-11:
firmware.ardupilot.org is down (servers are not available from June 8)
I watched another similar issue (1754). But it's now closed.

auturgy, 2019-06-11:
It’s available for me. Can you check again and paste the error message?

Regards,

James

dzotov2, 2019-06-11:
tracert (upper) and browser message (bottom).  I'm connecting from Russia Federation (tryed from separate machines and providers)

[image]

tridge, 2019-06-11:
definitely working from here

dzotov2, 2019-06-11:
strange behavior. Some devices have access to the site, some do not have it... Other resources working fine (http://www.ardupilot.org/, https://discuss.ardupilot.org/ )

auturgy, 2019-06-11:
It’s possible that a dns server somewhere along line is pointing to the old firmware server (due to caching). We upgraded to a more powerful build server last week, so the underlying IP address has changed (although the URL hasn’t).
Unfortunately firmware.ardupilot.org isn’t directly reachable via the server IP.
The website is definitely up and working from other places around the world.
I’m not a guru on this stuff so can’t really help more than that.
If you need a specific firmware, perhaps open a post on the forum asking for it.

I’d like to close this issue, but we’ve noted that changing the server hasn’t properly propagated everywhere.

Thanks for reporting this.

James

closed 2019-06-11 by dzotov2

peterbarker, 2019-06-11:
That IP address that @dzotov2 reported is the current one.

So... has that IP address been blacklisted by something/someone?  I guess we'll have to keep an eye on this.

dzotov2, 2019-06-11:
> That IP address that @dzotov2 reported is the current one.
> 
> So... has that IP address been blacklisted by something/someone? I guess we'll have to keep an eye on this.

Checked from different machines, stations, providers and conditions. Local\regional blocking of the IP \ domain http://firmware.ardupilot.org was also not detected. Some devices have access to the site, some do not have it.

peterbarker, 2019-06-11:
Are the traceroutes the same from different networks?

Paste a working traceroute, perhaps?

dzotov2, 2019-06-11:
for example, working tracert

[image]

+ second another

Трассировка маршрута к mega2.ardupilot.org [159.65.186.182]
с максимальным числом прыжков 30:

  1    <1 мс    <1 мс    <1 мс  serv.office2.naumen.ru [192.168.212.1]
  2    <1 мс    <1 мс    <1 мс  10.0.12.254
  3     5 ms     5 ms     5 ms  62.181.37.17
  4    19 ms    12 ms     5 ms  mow-b4-link.telia.net [62.115.161.120]
  5   122 ms   122 ms   121 ms  s-bb3-link.telia.net [62.115.116.232]
  6   121 ms   120 ms   123 ms  kbn-bb3-link.telia.net [62.115.139.169]
  7   117 ms   117 ms   118 ms  nyk-bb3-link.telia.net [213.155.134.50]
  8   122 ms   119 ms   118 ms  nyk-b3-link.telia.net [62.115.140.223]
  9                     *     Превышен интервал ожидания для запроса.
 10                     *     Превышен интервал ожидания для запроса.
 11   117 ms   120 ms   117 ms  159.65.186.182