ArduPilot/ardupilot_wiki #1808: Object Avoidance

modesto-c-morais, 2019-06-22:
[Object Avoidance](http://ardupilot.org/copter/docs/copter-object-avoidance.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

- Is it possible to maintain the object avoidance mode active under automatically controlled flight mode?
Regards,
Modesto

rmackay9, 2019-06-23:
@modesto-c-morais,

We are working on adding object avoidance to Rover and Copter's Auto, Guided and RTL modes.  Hopefully we can start beta testing of Rover-3.6 and Copter-3.7 in about 6 weeks and then you could give it a try.

closed 2019-06-23 by rmackay9

modesto-c-morais, 2019-06-24:
Hi Randy,

Please, soon you have that option, let me know.

Thank you,

Modesto

Modesto Morais
Responsável pela Inovação, Investigação e Desenvolvimento
Inovação, Investigação e Desenvolvimento
+351 229570029 | +351 963929125

[https://www.iep.pt/images/logo_iep.png]<http://www.iep.pt>     PORTO:
Rua de S. Gens 3717, 4460-817 Custóias
T. +351 229 570 000 | F. +351 229 530 594

LISBOA:
Polo Tec. de Lisboa, R. António Champalimaud Lt.1, Ed. CID 1.° Piso, Sala 103, 1600-546 Lisboa
T. +351 214 717 250 | F. +351 214 717 252

[https://www.iep.pt/images/fundo_iep.png]<http://www.iep.pt>
Saiba mais em iep.pt<http://www.iep.pt>

Enviada: 23 de junho de 2019 03:01
Para: ArduPilot/ardupilot_wiki <ardupilot_wiki@noreply.github.com>
Cc: Modesto Morais <CM@iep.pt>; Mention <mention@noreply.github.com>