ArduPilot/ardupilot_wiki #1811: Radio Control Calibration in Mission Planner

RavindraJadhav-dev, 2019-06-25:
[Radio Control Calibration in Mission Planner](http://ardupilot.org/planner/docs/common-radio-control-calibration.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

RavindraJadhav-dev, 2019-06-25:
In my quadcopters settings in mission planner, while transmitter calibrating my throttle key is showing rolle and right key showing yaw, how to swap them?

brunoolivieri, 2019-07-05:
Hello @RavindraJadhav-dev  did you tried to check your radio mix or the radio channel mapping?

Could you check each channel using only the radio and a servo or a motor with an each?

This does not look like a Wiki problem. I believe that you could discuss that on the Ardupilot forums: https://discuss.ardupilot.org/

Could you close this issue?

closed 2019-07-05 by Hwurzburg