ArduPilot/ardupilot_wiki #1837: TBS CROSSFIRE BLUETOOTH OUTPUT MAVLINK - from a complete newbie

arjrj16, 2019-07-11:
Hi all,
I would like to build an antenna tracker with a tbs crossfire. I am currently deciding between a ghetto station and Ardupilot. I have a tbs crossfire which can output Mavlink through Bluetooth, so I was wondering how I  would be able to use this with Ardu Pilot. Also, do I have to connect my video receiver to the flight controller like it is shown on the Ardu Pilot website - I would like to build a ground station for a rapid-fire receiver which I do not think can be connected up to something like that. Also, it is 5.8 ghz so I do know if this would really work. So far, I am running a dalrc f405 flight controller for my mini quad, which I will install a GPS and inav on. If I am correct, my tbs crossfire will then be able to output mavLink as Bluetooth. I don't know how I would do the rest with ArduPilot, but I assume it would go along the lines of wiring a Bluetooth module up to a flight controller that supports ardupilot. What is the difference between ArduPilot and Ghettostation? Would Ghettostation work better? I am pretty new to this, so sorry if I got any of this wrong. 
Thanks,
Arjun M

rmackay9, 2019-07-11:
Hi there.  Thanks for considering AP.  This issues list is for recording to-do items for the AP wiki so I'm going to close this issue.  It's best to create a new thread in the [antenna tracker support forum](https://discuss.ardupilot.org/c/antennatracker).

I'm afraid I know very little about inav or ghettostation but our tracker wiki is here: http://ardupilot.org/antennatracker/index.html

Normally the vehicle's flight controller is running AP outputting mavlink over some kind of telemetry radio and then there is another flight controller on the AP antenna tracker that has the other end of the telemetry radio.

closed 2019-07-11 by rmackay9

arjrj16, 2019-07-12:
Oh ok. Sorry about that - I am very new to github.