ArduPilot/ardupilot_wiki #1848: Auxiliary Function Switches

Tonynewton, 2019-07-18:
[Auxiliary Function Switches](http://ardupilot.org/copter/docs/channel-7-and-8-options.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
I am trying to get autotune working.  I am unable to access the switches from the Extended Tuning tab but the RC7Opt and RC8Opt buttons are grey and don't respond to the mouse pointer.
In the Full Parameter List tab, I was able to set the AUTOTUNE_AXES TO 1 (for roll tuning only).  I set AUTOTUNE_AGGR to 0.1 and AUTOTUNE_MIN_D to  0.001  
The copter did the roll tuning and beeped when it was done.  I left switch 7 high, landed and disarmed.  The new PIDs were not saved.  Does this have anything to do with the grey fields in the Extended Tuning Tab.  How does one activate the RC7Opt and RC8Opt buttins?

rmackay9, 2019-07-19:
This is a [MP display issue](https://github.com/ArduPilot/MissionPlanner/issues/2070) which is on the to-do list to fix before Copter-3.7 goes out.  I have to ask though why is Copter-3.7 being loaded onto the vehicle?  The latest stable is 3.6.9 and we highly recommend sticking with this version instead of the development version which hasn't undergone any beta testing yet.

As a side note, issues should really be raised first in the [support forums](https://discuss.ardupilot.org/c/arducopter) so I'll close this issue.  Thanks in any case for using AP!

closed 2019-07-19 by rmackay9

Tonynewton, 2019-07-19:
Hello Randy,Thank you for replying to my issue.  I am new to this and didn't realize that I was supposed to use the support forum.  In future, I will use the support forum.You people are doing a wonderful job with this project.  I have been able to build a working quad from scratch.  I will persist with the autotune and I know that I will eventually get it to save.  I will reload version 3.6.9 and use it.  Thank you for taking the time to help me.Best regards,Tony NewtonSent from my Galaxy Tab® A

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ArduPilot/ardupilot_wiki/issues/1848?email_source=notifications\u0026email_token=AJC7FPB3DBFEA6J5DQGICELQAEC7PA5CNFSM4IE6BZ22YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2KGVOA#issuecomment-513043128",
"url": "https://github.com/ArduPilot/ardupilot_wiki/issues/1848?email_source=notifications\u0026email_token=AJC7FPB3DBFEA6J5DQGICELQAEC7PA5CNFSM4IE6BZ22YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD2KGVOA#issuecomment-513043128",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]