ArduPilot/ardupilot_wiki #1866: re-organise common autopilots page

rmackay9, 2019-07-30:
As part of merging this PR:https://github.com/ArduPilot/ardupilot_wiki/pull/1863 we thought that a good way to display the autopilot hardware options would be to put all them in a table in alphabetical order (columns could be manufacturer, board, etc) and then add a column to show the state of open-ness (open, schematics available or closed)

@auturgy would you be interested in helping us with this?

auturgy, 2019-07-30:
Can do. Sorry I’m not on the call today.

Hwurzburg, 2019-11-01:
We've alphabetized already....for schematics/gerbers, I propose a link in each board page to its schematics there, rather than adding another page or xref...if a board archives you dont have to worry about updating another page..

Hwurzburg, 2019-11-14:
done and updated...

closed 2019-11-14 by Hwurzburg