ArduPilot/ardupilot_wiki #1869: Resolve wiki page build issues

rmackay9, 2019-08-02:
When building the wiki pages if seems like there are many issues displayed and some of them are new-ish.

Just one example:

common-appendix.rst:11: WARNING: toctree contains reference to nonexisting document u'docs/common-training-centers'

brunoolivieri, 2019-08-02:
I hope that I can try to help with those warnings from next week.

khancyr, 2019-08-02:
Ideally, we should add travis to check for new warning

brunoolivieri, 2019-08-02:
I understood that it would be a task to find errors in pages and fix them based on the warnings.

khancyr, 2019-08-02:
To have the warning, you just have to build the wiki, the warning will be displayed on the terminal.

brunoolivieri, 2019-08-02:
Yes, I was wondering to build, grep some warnings to try fix on the .rst files: https://pastebin.com/X8DFYCmp.

I did not get how Travis could help in this case.

brunoolivieri, 2019-08-05:
I did a quick check, and there were 466 warnings while building the wiki:

1) 324 warnings from "undefined labels"  (~70%)
2) 93 warnings from documents that are not in any toctree (~20%)
3) 35 warnings from parameters.rst regarding "duplicate explicit"  (~8%)

Regarding 1) This look hard to crosscheck. Is there an easy way to find the missed reference?

I mean: ACRO_EXPO is missing on copter/accro_mode.rst. Would be the better solution to remove such :ref:?

Regarding 2) Does it needs to check one by one? Is it just an alarm do try to show lost pages? I am asking it because of lots of files in this situation are easily accessible.

Regarding 3)  I believe that could be removed at once on the build scripts?

rmackay9, 2019-08-06:
ACRO_EXPO has been split into ACRO_RP_EXPO, ACRO_Y_EXPO

brunoolivieri, 2019-08-08:
I did a draft to investigate the parameters that may have had their names changed. So, the 324 pages can be summarized to the 66 labels below. Before read each page I compared the list with [parameters page](http://ardupilot.org/copter/docs/parameters.html) to get some clues. 

@rmackay9 Please, could you check if this is what you mentioned?

| Missing label                                                    | Looks like a parameter?                                                                          | TO-DO                                                                                       |
|------------------------------------------------------------------|--------------------------------------------------------------------------------------------------|---------------------------------------------------------------------------------------------|
| _atc_rate_ff_enab                                                | Is it ATC_RATE_FF_ENAB?                                                                          |                                                                                             |
| acro_expo                                                        | Yes!                                                                                             | Change the label. Note: "we have split it into two parameters ACRO_RP_EXPO and ACRO_Y_EXPO" |
| adsb_list_size                                                   | Is it ADSB_LIST_MAX?                                                                             |                                                                                             |
| avd_f_action                                                     | Is it AVD_F_RCVRY?                                                                               |                                                                                             |
| brd_safety_enable                                                | Is it BRD_SAFETYENABLE?                                                                          |                                                                                             |
| channel-output-functions_flaperon1_and_flaperon2                 | no?                                                                                              |                                                                                             |
| checking the motor numbering with the mission planner motor test | no?                                                                                              |                                                                                             |
| connect-escs-and-motors                                          | no?                                                                                              |                                                                                             |
| elevon_output                                                    | no?                                                                                              |                                                                                             |
| fs_batt_enable                                                   | Is the BATT_FS_LOW_ACT (set as zero) the newer vesion of this parameter?                         |                                                                                             |
| fs_batt_mah                                                      | Is the BATT_LOW_MAH the newer versions of this parameter?                                        |                                                                                             |
| fs_batt_voltage                                                  | Are the BATT_LOW_VOLT, BATT_CRT_VOLT, and BATT_FS_VOLTSRC the newer versions of this parameter?  |                                                                                             |
| guide-flaperons                                                  | no?                                                                                              |                                                                                             |
| invertedflt_ch                                                   | Are optinals channels used with value 43 the newer approach for this parameter?                  |                                                                                             |
| lgr_servo_deploy                                                 | Is the LGR_STARTUP the newer parameter?                                                          |                                                                                             |
| lgr_servo_rtract                                                 | Is the  LGR_STARTUP the newer parameter?                                                         |                                                                                             |
| manual tuning of roll and pitch                                  | no?                                                                                              |                                                                                             |
[... cut ...]
| learn_ch                                                         | Compass learn? Throttle learn? Temperature learn?                                                |                                                                                             |
| mag_enable                                                       | Would be COMPASS_ORIENT or COMPASS_EXTERNAL?                                                     |                                                                                             |

Hwurzburg, 2019-08-08:
Bruno. I am working on the errors for plane specific stuff right now...
fyi, many labels are deprecated but still need to be there to support older versions, so they wont have valid refs....I just eliminate the ref, but keep the param label if its obsolescence isn't horribly ancient....like INVERTEDFLT_CH

brunoolivieri, 2019-08-08:
Hi Henry,

Nowadays the wiki just present last Ardupilot repo parameters list. There is an Issue asking to maintain a version for each stable release, but I believe that will not be done short.

I am afraid to maintain some old parameters listed by removing the refs without any leave any documentation or indicative of a deprecated information. 

I guess that the build warnings are the only indicative of old information about parameters that we have. Without this warning in the build I am afraid that could be lost. 

Maybe would be better flag it to next wiki call that we will be present. Let me know your thoughts.

brunoolivieri, 2019-08-21:
@rmackay9 , please, 

MAG_ENABLE used to be an old parameter?

rmackay9, 2019-08-21:
@brunoolivieri, yes, it's been zapped somewhat recently (in the last few months) so it probably still exists in the stable versions but it will be gone in the next stable versions which should go out over the next few months.

Hwurzburg, 2019-09-11:
Closing now....all wiki build problems (except the code block error which I will continue to work on, and the delayed merge items) are now resolved....

closed 2019-09-11 by Hwurzburg