ArduPilot/ardupilot_wiki #1870: Cameras & Gimbals: rationalise 4 pages of setup instructions

rmackay9, 2019-08-02:
The [Cameras and Gimbals](http://ardupilot.org/copter/docs/common-cameras-and-gimbals.html) and [Mission Planning](http://ardupilot.org/copter/docs/common-mission-planning.html) landing pages contains links to several pages that describe how to setup and trigger a camera to take pictures from the pilot's transmitter and during missions.

There is a lot of repeated information on these pages so we should attempt to restructure the pages so that we reduce this duplication.  I think we can:

- Camera and Gimbals pages should focus on the hardware setup and ArduPilot parameter changes required to allow ArduPilot to trigger a camera using a specific piece of hardware (i.e. a seagull IR trigger, a DrotagX, etc).  These pages should have links to the Mission Planning pages so users can figure out the next step which is to actually trigger the camera in a mission or from the transmitter
- Mission Planning pages should focus on how to create missions that take pictures at the correct intervals, how to download the data and turn that into maps, etc.  This page or pages should have links to the "Camera and Gimbals" landing page section where users can find out the list of supported hardware devices they can use.

Mission Planner:

- [Camera Control and Auto Missions](http://ardupilot.org/copter/docs/common-camera-control-and-auto-missions-in-mission-planner.html)
- [Geotagging Images with Mission Planner](http://ardupilot.org/copter/docs/common-geotagging-images-with-mission-planner.html)

Cameras & Gimbals:

- [Camera Triggering Directly from AUX Ports](http://ardupilot.org/copter/docs/common-pixhawk-camera-trigger-setup.html)
- [Camera Triggering Configuration](http://ardupilot.org/copter/docs/common-camera-shutter-with-servo.html)
- [Camera Triggering using CHDK](http://ardupilot.org/copter/docs/common-apm-to-chdk-camera-link-tutorial.html)
- [Camera Triggering using CHDK Tutorial (non-standard integration)](http://ardupilot.org/copter/docs/common-chdk-camera-control-tutorial.html)

Use-Cases and Applications:

- [3d Mapping](http://ardupilot.org/copter/docs/common-3d-mapping.html)

Naterater, 2019-08-15:
Please notify me in these PR's as I would like to review/edit. I have a lot of experience with triggering, however little time to start this consolidation process. I agree it definitely needs work.

rmackay9, 2019-08-17:
@Naterater, great, thanks.  I was hoping to get your input on these pages actually once I get there.