ArduPilot/ardupilot_wiki #1875: Copter: add missing motor diagrams

rmackay9, 2019-08-05:
We are missing the following motor diagrams:

- Hexacopter H.  This is the same as Hexacopter X except motors 1 and 2 are closer to the vehicle's center
- BiCopter

rmackay9, 2019-09-02:
@lvale told me today that there are others missing include DJI motor orders

auturgy, 2019-09-02:
Yes, and betaflight motor order

Regards,

James

Hwurzburg, 2019-09-30:
add reverse tricopter

IamPete1, 2019-09-30:
to expand, Motor 1 and 2 remain on the right and left respectively as with standard tricopter. the tail motor 4 is now at the front https://github.com/ArduPilot/ardupilot/pull/10846

IamPete1, 2019-10-04:
not copter but we should also add diagrams for the rover omni frames

rmackay9, 2019-10-04:
@IamPete1, we have at least some of the rover omni frame here I think: http://ardupilot.org/rover/docs/rover-motor-and-servo-connections.html#omni-vehicles

IamPete1, 2019-10-05:
I was looking at the second page, so missed them completely, in any-case were missing the 4 inline mecanum wheels that I think we support?

rmackay9, 2019-10-07:
@IamPete1, no, I think that's one of the existing omni vehicles... maybe the OmniX.  I'm not totally sure but I don't remember us having to make any code changes for the person that got the omni rover working.

rmackay9, 2019-10-29:
We are also missing these for Quad:

- BetaFlight X
- DJI X
- CW X

Hwurzburg, 2019-11-10:
@rmackay9  if you can hand sketch, and send me a photo, I will work on these....could save us some $$ right?

rmackay9, 2020-01-27:
FYI the DJIX configuration came up today in this support thread: https://discuss.ardupilot.org/t/copter-4-0-1-released/51505/19

rmackay9, 2020-02-26:
This is the link to where we store the configurations: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp#L498