ArduPilot/ardupilot_wiki #1896: Common: add SPL telemetry link to our Telemetry landing page

rmackay9, 2019-08-12:
I think we should add a link to our Telemetry landing page that points users to the SPL solution for 3G/4G (or whatever it is!) telemetry.  [There is a blog post here about it](https://discuss.ardupilot.org/t/stretching-comm-links-from-indoors-to-the-globe/45896) and we've already done the same for a competing product called UAVCast.

auturgy, 2019-08-12:
Yup. FYI SPL is an Iridium satcom datalink

adnan-ulhaque, 2019-08-27:
NAT Traversal through telecomm / sattelite service provider is the key here. I believe this an extremly important topic to connect telemetry+fpv over bvlos (beyond visual line of sight). UAVcast and other alternatives are not free . There are free public cloud services where Raspberry Pi (with Pixhawk) can be connected to MP on TCP via any data cellular SIM e.g. below...

https://discuss.ardupilot.org/t/fpv-over-internet-via-rpi3/22597?u=adnan_ul_haque

Hwurzburg, 2019-11-01:
PR# 2178 addresses this

closed 2019-11-10 by Hwurzburg