ArduPilot/ardupilot_wiki #1962: ardupilot airsim sitl instructions edit needed

UASOP, 2019-09-02:
I was working with the instruction today and ran into an issue not just from an existing setup for airsim with ardupilot but also from restarting the setup again on another computer. It seems as though the settings.json text for arducopter requires a slight edit otherwise an error occurs with the settings text that is provided on the wikipage. The wikipage currently provides:

 },
  "Vehicles": {
      "Copter": {
          "VehicleType": "ArduCopter",
          "UseSerial": false,
          "AllowAPIAlways": false,
          "UdpIp": "127.0.0.1",
          "UdpPort": 9003,
          "SitlPort": 9002
        }
    }
}

but this results in an incompatible vehicle type warning when running airsim  on windows 10, as shown in the screen capture below. 
[image]

I did some checks and by setting the "V" in ""Vehicles" to a **Lower Case so that is becomes "vehicles"** the issue was resolved. Hope this helps a few others who might have run into this small issue. 

It would be awesome if it could be changed in the Wikipage (http://ardupilot.org/dev/docs/sitl-with-airsim.html) as well, if it helps avoid heartache during setup for others. 

Thanks and have a good day.

rajat2004, 2019-09-02:
Hi @UASOP, 
Is the AirSim installation on Windows upto date with the current master branch? If not, then that would be the cause of the error since the support was merged very recently and there hasn't been a new release of AirSim after that.

I just tried running the simulation with the settings given in the Wiki and everything was working correctly. Is the `SimMode` field in the `settings.json` set to `Multirotor`?
Please send the settings which you're using if the issue still persists.

UASOP, 2019-09-02:
Thank you for the rely @rajat2004.

 I was just looking through the cpp files and then checked it in comparison to the master branch. I am using the master branch. Last time I updated the branch was 7 days ago, so seeing that the latest merge for the arducopterapi.cpp was six days ago would make since. 

I am going to try to rebuild the master branch and post findings here and follow up as soon as possible. Thanks again!

UASOP, 2019-09-02:
Thank you for the rely @rajat2004.

 I was just looking through the cpp files and then checked it in comparison to the master branch. I am using the master branch. Last time I updated the branch was 7 days ago, so seeing that the latest merge for the arducopterapi.cpp was six days ago would make since. 

I tried rebuilding the master branch about an hour ago and and tested the settings.json file in windows, I am sorry to say that I am still receiving the same error. The the entire settings file is below. 

{
  "SettingsVersion": 1.2,
  "LocalHostIp": "192.168.1.13",
  "LogMessagesVisible": true,
  "SimMode": "Multirotor",
  "OriginGeopoint": {
    "Latitude": -35.363261,
    "Longitude": 149.165230,
    "Altitude": 583
  },
  "Vehicles": {
      "Copter": {
          "VehicleType": "ArduCopter",
          "UseSerial": false,
          "AllowAPIAlways": false,
          "UdpIp": "192.168.1.15",
          "UdpPort": 9003,
          "SitlPort": 9002
        }
    }
}

rajat2004, 2019-09-02:
That's very strange
I tried using the same settings as yours, working correctly on my machine. I'm on a Linux machine right now, but it shouldn't be any different. Will test on Windows in some time

Just to confirm, you're still getting the same error about the `incompatible vehicle type`?

UASOP, 2019-09-02:
@rajat2004 correct, it is still showing the incompatible vehicle type error. Thanks for looking into it :)

rajat2004, 2019-09-02:
@UASOP I checked it on Windows as well and AirSim worked properly, seems it's time to dig deeper

From what you mentioned earlier that you have updated your local repo, this mostly isn't the problem, but please send the latest commit id in your local AirSim repo, you can do `git log` and the first commit in the list will be the one needed
Mine is [622a9d7e15c5489e6aaacd81db283619380b38c4](https://github.com/microsoft/AirSim/commit/622a9d7e15c5489e6aaacd81db283619380b38c4) (Latest master as of today)

What version of Unreal Engine are you running?
Also check if PhysXCar and SimpleFlight are working correctly

UASOP, 2019-09-02:
@rajat2004  Good to know it is working on your end. 

My latest when using git log is 622a9d7e15c5489e6aaacd81db283619380b38c4. I have tried both the car and the simpleflight and they are both working properly.

rajat2004, 2019-09-02:
This is really very strange since a few other people have tried out ArduCopter with AirSim without any problems.

You could try doing a clean rebuild of AirSim, maybe that would fix it
Also, could you share the `Saved` folder in the project, will be inside `AirSim/Unreal/Environments/Blocks/`, that contains logs and could help solving this

UASOP, 2019-09-02:
@rajat2004 I agree this is strange. I have tried deleting the folder for airsim where it is built on my computer and then downloading it using the git clone command and then running the build.cmd command as well. Whats another way to do a clean rebuild?

UASOP, 2019-09-02:
The only thing I haven't tried full is the clean_rebuild.bat script. I will try that now.

rajat2004, 2019-09-02:
@UASOP yes, please try running the `clean_rebuild.bat` script, let's see if that helps
Also, after updating the local repo, did you run the `update_from_git.bat` script from inside the Blocks folder? I sometimes forget to do that

UASOP, 2019-09-02:
@rajat2004 i performed both commands in their respective folders. I tried running the blocks environment in Unreal engine throught the VS project file and when I hit play, with the settings.json file set with the code provided in the tutorial, unreal engine just froze. In the saved folder it showed a crash report. I have attached the log files below:
[Blocks.log](https://github.com/ArduPilot/ardupilot_wiki/files/3566974/Blocks.log)
[unrealversionselector-2019.09.02-12.08.05.log](https://github.com/ArduPilot/ardupilot_wiki/files/3566975/unrealversionselector-2019.09.02-12.08.05.log)

rajat2004, 2019-09-02:
@UASOP Ahh, finally something other than the incompatible vehicle error!
If the Unreal Engine just froze, then the most likely cause is that the ArduPilot SITL is not running since, in the log, it can be seen that the Copter vehicle has started. Try running the Ardupilot SITL as well and see if it connects

There's another error visible in the top of the log file about the `.dll` files not found, haven't seen this before but from what I found on net, you could try deleting the Binaries, Intermediate, Saved folders in the Blocks environment and rebuild again

Hope this helps

UASOP, 2019-09-02:
"Smacks myself in the forehead, while smiling with releif" 

@rajat2004 thank you!! SITL up in bashonwindows and then I hit play on the blocks windows project, boom, it works. Now I understand.

So in order to have  unreal working the environment has to be run from the unreal engine project file by pressing play. I am going to explain below what I did wrong to get the incompatible vehicle error so if someone else stumbles amongst this same issue they can be able to reference this issue thread and hopefully figure it out as well. 

I was originally able to use each individual environments folder file with the run.bat or unreal.exe for the environment. screen cap shown below of the environment folder. 
[image]

The executables in the environment folders worked with Arducoptersolo, car, PX4Multirotor, and simpleflight in the setting.json file, but with the arducopter settings.json I would receive the vehicle incompatible error. So in order to run the arducopter in environments they need to be started from the unreal engine project file only. Since the individual environment folders available for download here (https://github.com/Microsoft/AirSim/releases) keep giving the incompatible vehicle error if either the run.bat or environment.exe are run directly from those folders.

UASOP, 2019-09-02:
This now brings the question. How does those individual environment folder know to work with the detailed vehicles such as Arducoptersolo, car, PX4Multirotor, and simpleflight in the setting.json file, but not the arducopter SITL setup?

rajat2004, 2019-09-02:
@UASOP Really glad to see that it's working correctly now!

> This now brings the question. How does those individual environment folder know to work with the detailed vehicles such as Arducoptersolo, car, PX4Multirotor, and simpleflight in the setting.json file, but not the arducopter SITL setup?

That's because support for Arducopter is still only in the master branch, they haven't released a new version with this included (hopefully it happens soon!)

> So in order to run the arducopter in environments they need to be started from the unreal engine project file only. Since the individual environment folders available for download here [...]

Just to clarify, from the image you posted above, it seems that you were using the release version of AirSim environments rather than the Blocks? And going for the Blocks or any other non-release environment worked?
Or were you somehow able to use the release environments with Ardupilot?

UASOP, 2019-09-02:
> Just to clarify, from the image you posted above, it seems that you were using the release version of AirSim environments rather than the Blocks? And going for the Blocks or any other non-release environment worked?
Or were you somehow able to use the release environments with Ardupilot?

@rajat2004 Yes I was using released versions of the environments. I now understand that the latest releases with arducopter included are not out yet, thanks for the clarification on that. 

It only worked with non-release environments, such as Blocks and now Landscape mountain. I am now trying to build a package for it for Unreal.

UASOP, 2019-09-02:
The environment package build is finished and working on windows with the environment.exe file!

rajat2004, 2019-09-03:
@UASOP Nice to know that it's all working correctly now
Based on this long discussion, seems like there can be some improvements to the Wiki page.

Few things I'm thinking of are-
1. Make the requirement of building AirSim from source and not using the release environments more explicit
2. Give more info or links on how to debug and clean rebuild AirSim, especially on Windows

Another which I'm thinking of is having some kind of index of the topics in the page itself since it's quite long currently, but not exactly sure how to do it, will look into this!

Any other points where it could be improved?

rajat2004, 2019-09-05:
I have opened https://github.com/ArduPilot/ardupilot_wiki/pull/1977 for some improvements to Wiki page

Additionally, @UASOP since the problem has been solved, I think this issue can be closed. 
Do comment or open a new issue if other problems pop up!

UASOP, 2019-09-08:
Certianly @rajat2004 Thank you again for all of your help and the work that you have put into getting ardupilot SITL working with AirSim!

closed 2019-09-08 by UASOP