ArduPilot/ardupilot_wiki #1966: Motor and Servo Configuration Failed

CrisbenDev, 2019-09-02:
[Motor and Servo Configuration](http://ardupilot.org/rover/docs/rover-motor-and-servo-configuration.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I'm configuring a rover. I'm trying to set the parameters for "Ground Steering", the problem is that the servo doesn't work. I want it to turn the wheel. It just doesn't respond.
I already checked the control (spektrum dx7s) it's working fine.
Acctualy, if I change the parameter "Ground Steering" to "RC Pass Thru" or "RCIN1" it works. But I don't think its the right thing to do.
Can someone help me with that please?

rmackay9, 2019-09-02:
@CrisbenDev, it's best to raise support issues in the [Rover forum here](https://discuss.ardupilot.org/c/ardurover/rover-35).  Also include a [dataflash log](http://ardupilot.org/rover/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html) file as well if possible.

My initial guess is that the vehicle is in Hold mode instead of something like Manual or perhaps the vehicle hasn't been armed.  Anyway, best to post in the forums.

closed 2019-09-02 by rmackay9

CrisbenDev, 2019-09-04:
I'm greatfull for your help :D