ArduPilot/ardupilot_wiki #1975: Setting up the waf Build Environment on Windows using Cygwin - use python27-future, because python2-future is not available anymore

WillyZehnder, 2019-09-05:
[Setting up the waf Build Environment on Windows using Cygwin](http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html)
The recommended commandline: "setup-x86_64.exe -P autoconf,automake,ccache,gcc-g++,git,libtool,make,gawk,libexpat-devel,libxml2-devel,python27,python2-future,python27-libxml2,python27-pip,libxslt-devel,python27-devel,procps-ng,zip,gdb,ddd" is trying to install python2-future. But it fails with no comment, because that's is not available anymore. 
At building the "plane" target you get very early problems: 
[image]

So it's necessary to replace python2-future by python27-future in the document and in the corresponding ps1-file (https://github.com/ArduPilot/ardupilot/blob/master/Tools/environment_install/install-prereqs-windows.ps1)

Hwurzburg, 2019-11-01:
@peterbarker what is needed here and how should we accomplish it?

WillyZehnder, 2019-11-02:
@Hwurzburg : Maybe I'm not asked, but I will summarize my suggestions (sorry, I'm not so familiar with GitHub)
1) In Document http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html you should replace python2-future by python27-future
2) In the corresponding ps1-file (https://github.com/ArduPilot/ardupilot/blob/master/Tools/environment_install/install-prereqs-windows.ps1) you should replace python2-future by python27-future

peterbarker, 2020-09-07:
Should we instead be looking at killing cygwin as a supported mechanism for running ArduPilot?

Windows 7 is no longer supported by MicroSoft, and (FWIU!) wsl and wsl2 work far better.

... I might also point out that until last week I didn't actually have a working Windows machine, so assigning this sort of thing to me doesn't make a huge amount of sense :-)

Hwurzburg, 2020-09-07:
Peter, who should be involved in this? Pierre?

peterbarker, 2020-09-08:
Turns out that we can't kill cygwin - there's several processes that still work way better with it in existence.

@Hwurzburg I usually try to rope @stephendade in for MAVProxy and pymavlink things for Windows :-)

stephendade, 2020-09-08:
>Should we instead be looking at killing cygwin as a supported mechanism for running ArduPilot?

We still need Cygwin, as it's the only method to compile native SITL executables for Windows. This is what Mission Planner uses for it's simulation tab

For reference, the original issue was fixed some time ago when we migrated Cygwin to Python3. So we can close this issue.

closed 2020-09-14 by tridge