ArduPilot/ardupilot_wiki #1995: Gazebo with SITL Documentation Is Outdated

AhmedHisham1, 2019-09-09:
I want to publish the Gazebo’s virtual camera stream to a ROS topic. The documentation page says that this is done using the "ros_gazebo_camera" package. However, I can't find this package anywhere. Hence, I think this info. is outdated or false.

doc link: http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html

Is there any workaround to publish the virtual camera feed to a ROS topic?

I'm trying this with the default SITL "iris_demo" model.

brunoolivieri, 2019-09-10:
@AhmedHisham1  thanks for informing that information could be not accurate anymore.

Have you tried the [discussion forums](https://discuss.ardupilot.org) for help?

rajat2004, 2019-09-11:
Not an expert on ROS (actually haven't used it in some time!)
The name is probably incorrect, try looking for `gazebo_ros_camera` plugin, mentioned here - http://gazebosim.org/tutorials?tut=ros_gzplugins#Camera

I think you should be able to find some more info on how to accomplish this, this could be helpful - https://discuss.px4.io/t/how-to-add-a-ros-camera-to-iris-for-gazebo-simulation/5118
This is specific to PX4, but ROS-related info is still plentiful

Hwurzburg, 2019-11-01:
@peterbarker Peter, can you shed some light on this?...what is needed and who should do it or provide me the details to get it done...

khancyr, 2019-11-01:
I will update the doc

Hwurzburg, 2019-11-01:
Thanks Pierre

To-beas, 2021-07-11:
Hi,

Ist the documentation up to date by now? Following the doc for this I can start gazebo environment with plugin but I get an error on starting sitl frame gazebo-iris: 

$ ../Tools/autotest/sim_vehicle.py -f gazebo-iris --console --map

BUILD SUMMARY
Build directory: /home/tobias/ardupilot/build/sitl
Target          Text     Data    BSS     Total  
------------------------------------------------
bin/arducopter  3564783  157539  161576  3883898

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (5m18.753s)
SIM_VEHICLE: Using defaults from (/home/tobias/ardupilot/Tools/autotest/default_params/copter.parm,/home/tobias/ardupilot/Tools/autotest/default_params/gazebo-iris.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/tobias/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/tobias/ardupilot/build/sitl/bin/arducopter" "-S" "--model" "gazebo-iris" "--speedup" "1" "--defaults" "/home/tobias/ardupilot/Tools/autotest/default_params/copter.parm,/home/tobias/ardupilot/Tools/autotest/default_params/gazebo-iris.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--map" "--console" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501"
RiTW: Starting ArduCopter : /home/tobias/ardupilot/build/sitl/bin/arducopter -S --model gazebo-iris --speedup 1 --defaults /home/tobias/ardupilot/Tools/autotest/default_params/copter.parm,/home/tobias/ardupilot/Tools/autotest/default_params/gazebo-iris.parm -I0
[Run MavProxy] An exception has occurred with command: 'mavproxy.py --map --console --out 127.0.0.1:14550 --out 127.0.0.1:14551 --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501'
[Errno 2] No such file or directory

Every time though it starts compiling ~800 packages. Thanks for all the work so far.

srmainwaring, 2022-12-20:
Hi @Hwurzburg - the current wiki information for using Gazebo with SITL refers to a version that is now in maintenance (Gazebo11 or Gazebo classic) and uses plugins in external repos that are mostly maintenance only.

The current Gazebo version supported by OSRF is [Gazebo Garden](https://gazebosim.org/docs/garden/install). We have plugins to support this version in https://github.com/ArduPilot/ardupilot_gazebo and also a number of example models in https://github.com/ArduPilot/SITL_Models/tree/master/Ignition.

OSRF have just finished a major [namespace rename back to Gazebo from Ignition](https://community.gazebosim.org/t/a-new-era-for-gazebo/1356). Now that's complete I think it makes sense to revisit the ArduPilot projects and documentation bring them up to date. The main ArduPilot plugin repo is current with Gazebo Garden, and the project README reflects that. The subfolder and docs in the SITL_Models repo still refers to Ignition so I'll be updating that next.

I'd like to help to bring the wiki docs up to date and provide a good walk through the various examples. Gazebo is quite feature rich, not as good with aerodynamics as RF, but it does have good peripheral support and provides an environment for non-flying vehicles that might be better utilised if we documented it better.     

I'm happy to take your steer on structure etc. but think it would be good to have a meta-issue to outline a plan for updating the docs that can be tracked by a series of smaller issues / PRs.

Finally, many of the ROS examples could be replicated in simulation using Gazebo. It might be nice to have documentation to reflect this. My preference would be to target the most recent environment: ROS Humble / Gazebo Garden / Ubuntu 22.04. There is still a large community using ROS Melodic / Noetic with Gazebo11 - but that environment is becoming hard to support (and near impossible on macOS). I'd recommend prompting people to make the shift as the tools for ROS2 support are all available now (i.e. mavros is running on ROS2 and works well).

Hwurzburg, 2022-12-20:
@srmainwaring feel free to post an outline for tracking here and to make contributions in prs...you can use the present format for the Gazebo docs as a template, but feel free to deviate.....bullets can be done thusly
- [ ] first task
- [ ] second task

srmainwaring, 2022-12-20:
## Update Gazebo SITL Documents

Task list for updating the Gazebo SITL documents. Free free to suggest items to add...

### Legacy version - Gazebo11

- [ ] Add page for the previous version of Gazebo (Gazebo11) using the [existing instructions](https://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html).

### Landing page (Using Gazebo Simulator with SITL)

- [ ] Update preconditions - link to [Gazebo Garden install instructions](https://gazebosim.org/docs/garden/install).
- [ ] Update recommended versions: Ubuntu - Focal and Jammy; MacOS - Big Sur and Monterey.
- [ ] Update link to the ArduPilot hosted version of the plugin.
- [ ] Mirror the installation instructions from the README and the examples for the Iris and Zephyr.
- [ ] Add link to examples provided in SITL_Models.
- [ ] Record new video with using the new environment (where to post?). 
- [ ] Add links to some examples - perhaps the ship landing demo?

### Technical documents (for devs)

- [ ] Explain how the Gazebo plugin now uses the SITL - JSON backend. 
- [ ] Document other technical aspects of the core plugin - lockstep, control types, conversion of PWM to commands.

### Technical documents (for users)

- [ ] Add guide to adding a new vehicle.
- [ ] Add guide to adding peripherals.