ArduPilot/ardupilot_wiki #2019: Adding Last Updated Dates on Wiki Pages

Gone4Dirt, 2019-09-18:
I wanted to put this out there as an idea and see what peoples thoughts are.

As we update wiki pages in the future, can we start added a 'last updated' date at the bottom of the page.

Due to the speed of changes/development in ArduPilot,  sometimes the wiki doesn't represent the most up to date information and the information in the discourse threads can be more up to date.  If we simply added a date at the bottom of all future pages it would make it a lot easier to know if the information is valid or not.

Thoughts?

Hwurzburg, 2019-09-18:
I think that it would be a good thing...and possibly could be done automatically upon revision and merge....I'll bring it up in the next WIKI call

another issue is that we really can't update the WIKI on a new feature until its at least in Beta...often months after it appears in all the DEV versions in the "latest" folder...

brunoolivieri, 2019-11-26:
I have an idea of how to do it lightly. Let me do a shot.

brunoolivieri, 2019-11-27:
I did a script that produces that: http://www.olivieri.com.br/ardupilot/wiki_last_update_test/dev and http://www.olivieri.com.br/ardupilot/wiki_last_update_test/copter/

The last line in file presents the last commit date of itself.

What do you think?

Gone4Dirt, 2019-11-27:
That looks good.  Does what I was thinking of.  So the script you wrote automatically adds the date that the bottom?

brunoolivieri, 2019-11-28:
I just did a PR ##2278 that shows that.

brunoolivieri, 2019-12-04:
I close this one due to the discusses in #2278. Please feel free to re-open.

closed 2019-12-04 by brunoolivieri