ArduPilot/ardupilot_wiki #2025: Documentation: Unclear wording around Clock/Time Syncronisation

bromeara, 2019-09-20:
I am using a Jetson TX2 on an Auvidea 120 to communicate with a Pixhawk 4, and have used both the master branch and Arducopter 3.6.7 for this setup. We are currently experiencing extremely high mavros/timesync_status RTTs and intend to use the UAV indoors without a GPS system. So in debugging our RTTs(present both with and without GPS lock and still unsolved) problem without GPS, I noticed our flight controller would put out system time messages dated to 1/1/1970.

I then came across this bit of documentation in https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/ros-timesync.rst stating 

> The flight controller's time is set from the GPS by default but this can be modified to be set from the SYSTEM_TIME command by setting RTC_TYPES = 1 (for "MAVLINK_SYSTEM_TIME")

My interpretation of this was that one would set RTC_TYPES by sending a mavlink SYSTEM_TIME command and that the value of 1 represented MAVLINK_SYSTEM_TIME. However, the command only has 2 parameters neither labeled RTC_TYPES. I was able to find where the value was set, inside of the set_utc_usec function in AP_RTC, but I cannot find where that function is called from(grep -iRl returned too many instances to be helpful to me).

My second assumption was that RTC_TYPES would be set as a parameter, but the only documentation I could find on the Ardupilot site and looking through others *.param files was for a parameter called BRD_RTC_TYPES which seems to be for an external radio and not the serial port since the value defaulted to 1(MAVLINK_SYSTEM_TIME).

Finally I just manually appended the line RTC_TYPES,1 to my *.param file as a hail mary but the Pixhawk TIMESYNC message stayed the same. So I feel as I am missing something about this process. It was originally written by @rmackay9 so if you or anyone else would be able to shine some light on the meaning of this or how to get the Pixhawk to time sync from a non-GPS source I would be able to update the documentation.

rajat2004, 2019-09-20:
Maybe this could help - https://github.com/ArduPilot/ardupilot_wiki/issues/1932

bromeara, 2019-09-20:
Ok so I'm still learning the ins and outs of the code, but thank you for the links the reason I wasn't finding RTC_TYPES is that the parameter group names stack, so BRD_RTC_TYPES would be the parameter I'm looking for so this at least solves my question on the documentation.

closed 2019-09-20 by bromeara