ArduPilot/ardupilot_wiki #2045: Benewake TFmini / TFmini Plus lidar

adalta87, 2019-10-02:
[Benewake TFmini / TFmini Plus lidar](http://ardupilot.org/copter/docs/common-benewake-tfmini-lidar.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

This is wrong as the TFmini plus as far as I have been made aware of by the community is not able to use UART communication and as such needs to be expressed in the wiki as to not confuse potential buyers of this sensor that think it can be used on UART. I am aware that the TFmini has both I2C and UART commo's available but the plus version has yet to implement the uart commo. I was hoping this page could reflect that the tfmin supports both 12c and the tfmini plus only at this time supports i2c. considering these are different sensors they should at least have their own sections on setup even though the setup is almost identical to each other.

rmackay9, 2019-10-02:
yes, I agree that the tfmini and tfminiplus pages should be split.

peterbarker, 2019-11-01:
I have a uart-version TFMini Plus here, and it is working just fine and dandy on a UART, using the TFMini driver (`RNGFND1_TYPE=20` (BenewakeTFmini)).

I think "the community" may need poking to say that it does work on UART, you just need to specify the Mini driver, not the Plus driver.  Perhaps we should adjust the paramter description for the Plus to include "i2c" in it - that driver is i2c-specific.

adalta87, 2019-11-01:
That would make the most since especially considering plenty of people ask if you have read the wiki! Might as well have the correct information or its not really worth having at all. There have been plenty of folks complaining about this on the forum and its a bit counterintuitive if you have a setting for tfmini plus but dont specify that its for i2c only and probably should also provide on the tfmini mini plus setup that to use tfmin plus on uart then set to param 20 mot 25 or whatever the mini plus is. 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ArduPilot/ardupilot_wiki/issues/2045?email_source=notifications\u0026email_token=AFODXTZOVJFINKXQIZKRWUDQRN7JDA5CNFSM4I4Z7FXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECZWQSY#issuecomment-548628555",
"url": "https://github.com/ArduPilot/ardupilot_wiki/issues/2045?email_source=notifications\u0026email_token=AFODXTZOVJFINKXQIZKRWUDQRN7JDA5CNFSM4I4Z7FXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECZWQSY#issuecomment-548628555",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Hwurzburg, 2019-11-10:
#2212

closed 2019-11-10 by Hwurzburg