ArduPilot/ardupilot_wiki #2047: DShot and BLHeli ESC Support

adalta87, 2019-10-03:
[DShot and BLHeli ESC Support](http://ardupilot.org/copter/docs/common-dshot.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
Is this also the page that will be used for esc telemetry considering it shows a telem connection on esc which is not necessary for using bl_heli 32 passthrough or is there a different page for that, if this is not the page for esc telem then the image of the connection to telem needs to be removed and a separate page created for esc telem or another sub section on this page.  One thing to note that I have seen a lot of people having issues with esc telem and it not showing in the MP status tab is because the telem port they are connecting to the flight controller has not been set to mavlink 2 without mavlink 2 selected it will not show.

auturgy, 2019-10-03:
What is the specific issue? 
It’s not clear to me what you want changed.
This page clearly states it contains information on three aspects: DShot, ESC telemetry, and blheli pass through.

Mavlink 2 is the default for telemetry links: those users experiencing issues with it need to update the firmware on their telemetry radios, not change a parameter. Not receiving esc telemetry is a symptom (although there are many other mavlink 2 messages that the user will also not be receiving), and neither the cause nor solution are related to this page.

adalta87, 2019-10-03:
To start this page does not clearly define that it is also esc telemetry unless that is what dshot means. And i do not believe that mavlink 2 is the default setting, considering i just did a fresh install of 4.0 dev and it was clearly not the default protocol, this also has nothing to do with the radio they are using as i dont even use mavlink for my radios and use raw data so i cannot speak to folks using other radios than rfd900x. This page needs to explicitly have either 2 or 3 sections bl heli passthrough and esc telemetry "aka dshot" if thats what dshot means or bl heli pass through dshot and esc telem. Again i am no expert about what dshot blah blah blah means whether its dshot150-1200. Why is the telem photo connection in the same part with bl heli passthrough, considering you do not even need the telem connected to get passthrough. Telem connection is only for esc telemetry for example rpm,temp,current and volts and has nothing to do with passthrough. Maybe just you need to restructure how it is written but in its currrent state is not really a good wiki as it has things in sections that don't belong or is missing information. 

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/ArduPilot/ardupilot_wiki/issues/2047?email_source=notifications\u0026email_token=AFODXT54L7JEVWXTO77IKUDQMWJFTA5CNFSM4I4662JKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAHF7YY#issuecomment-537812963",

"url": "https://github.com/ArduPilot/ardupilot_wiki/issues/2047?email_source=notifications\u0026email_token=AFODXT54L7JEVWXTO77IKUDQMWJFTA5CNFSM4I4662JKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAHF7YY#issuecomment-537812963",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

vierfuffzig, 2019-10-03:
imho there's separate issues you're referring to:

1.    DShot is a protocol that's not necessarily related to BLHeli ESC firmware. having both topics on one wiki page might suggest this relation.

2.    the wiki page's recommendation to use DShot when using BLH passthrough might be misunderstood. from a multirotor perspective it is obviously recommended to use the best possible esc protocol available. it is not related to setting up BLH passthrough though. 

3.    sending BLH esc data via ardupilot telemetry requires using mavlink2. this is the default on the console, but not on the telemetry ports as for now. it might be worth adding to the page.

i agree the page's didactic structure might leave room for optimization with fellow users in mind that haven't setup BLH on ardupilot hundreds of times already.

adalta87, 2019-10-03:
I don't think adding mavlink 2 might be something you need to add but
should absolutely be something you add, how are people to know that, that
is what is needed unless it is specified in the how to. be like me sending
you a game to play without any instructions, totally worthless. Again if
the community is going to rely on the wiki then the wiki needs to be
informative not ohhhh you need to search through hours for threads to
hopefully find your answer. Again I think the structure of this page is
quite messy and needs to be cleaned up and more defined such as 3
subsections clearly labeled with bold for each new section *BL-Heli
Passthrough, DShot, and ESC Telemetry. * If I had the know how I would
start editing this but again I have read the wiki on how to edit the wiki
and still don't understand.

Hwurzburg, 2019-10-09:
PR merged for updated wiki page

closed 2019-10-09 by Hwurzburg