ArduPilot/ardupilot_wiki #2049: Complete Parameter List

adalta87, 2019-10-03:
[Complete Parameter List](http://ardupilot.org/copter/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Is the parameter list generated on stable releases or on the latest release, currently the AC full parameter list is missing servo_blh parameters.

adalta87, 2019-10-03:
found it

adalta87, 2019-10-03:
SERVO_BLH_PORT: Control port
Note: This parameter is for advanced users
This sets the telemetry port to use for blheli pass-thru

Using the word telemetry is quite confusing in this sense when you read all the wiki together for many people this parameter was set to what they have their esc telemetry serial port to which is wrong as bl_heli uses the pwm connections to connect a rewording of this to take out telemetry and use usb or serial would help greatly.

brunoolivieri, 2019-10-04:
Hello, 

The parameters page is auto-generated from the latest source of each vehicle. A  PR for versioning is being under analysis. 

Regarding the SERVO_BLH_PORT parameter description, do you mean to make explicit which port is used for blheli pass-thru?  In general, all documentation refers to the FC ports.

adalta87, 2019-10-04:
It would be nice if the servo_blh_port did not mention telemetry in the description because when you read that description and the way esc telem, bl heli, dshot wiki page reads using the word telemetry is confusing folks because they think that is the port they need to set to get esc telem. This isn't just the description of the param but also how the wiki read it is slightly confusing and I will also provide links in the forum where folks have set that port to what they have esc telemetry connected to. telemetry is telemetry but a direct passthrough to bl heli suite to me should not be called telemetry. Would be better named serial port or uart port to connect to bl_heli passthrough and also probably should note that as of current only direct usb is supported meaning console which is 0.

Hwurzburg, 2019-10-04:
I have a PR submitted for this

Hwurzburg, 2019-10-08:
PR merged

closed 2019-10-08 by Hwurzburg