ArduPilot/ardupilot_wiki #2055: I have observed multiple instances of "unhealthy pose" and cannot find the details of this in the ardupilot documentation.  Does anyone have it?

Csweeten, 2019-10-09:
[Search](http://ardupilot.org/planner/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Hwurzburg, 2019-10-09:
you will need to supply the information requested when entering an issue for this to be addressed...ie firmware, vehicle, etc.

Csweeten, 2019-10-09:
Pixhawk with firmware 3.6.10 (also observed on 3.6.9) and helicopter (single rotor)

rmackay9, 2019-11-09:
ArduPilot doesn't output a message like this and I don't think I've ever seen it personally.  It's probably coming from the ground station itself, perhaps based on a mavlink message sent from AP.  It's possible that we can trace it back to what it means in ArduPilot if we have a dataflash or tlog and some information about what ground station is being used.

In a general sense "pose" means attitude.  So "unhealthy pose" would mean that the vehicle has a bad attitude estimate.  This is very rare but could happen if there are very bad vibrations.

I think I will close this for now at least .. I think it is important that we tell users about the meanings of error messages, but because this isn't coming from AP as far as I know.. I might close it.

closed 2019-11-09 by rmackay9