ArduPilot/ardupilot_wiki #2057: Pozyx for Non-GPS Navigation

vdacian, 2019-10-10:
[Pozyx for Non-GPS Navigation](http://ardupilot.org/rover/docs/common-pozyx.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

vdacian, 2019-10-10:
did n find pozyx config parameters in Mission Planer.

Can u help?

Thanks

Hwurzburg, 2019-10-10:
You might have better results asking in the Copter or Mission Planner support forum at discuss.ardupilot.org...... this is for Wiki documents issues and changes...

closed 2019-10-10 by Hwurzburg

rmackay9, 2019-10-11:
@vdacian, I think the most likely cause is that the pixhawk flight controller suffers from the 1MB flash limit which means that it uses the "px4-v2" or "fmuv2" firmware (depending upon whether it's NuttX or ChibiOS).  In any case, as @Hwurzburg says this is the wiki to-do list so probably best to raise this in the support forums.