ArduPilot/ardupilot_wiki #2058: GCS Failsafe

Sir-Lamba, 2019-10-10:
[GCS Failsafe](http://ardupilot.org/copter/docs/gcs-failsafe.html) - 
Hello ardupiloters :grin:
I'm a new bee, carrying out a research on UAV failsafe systems.
Can GCS Failsafe be set on other GCS softwares aside Mission Planner?

brunoolivieri, 2019-10-10:
Hi there,

This is the wiki to-do list. Probably best to post questions in the forums. https://discuss.ardupilot.org/

Regarding the failsafes, did you checked the parameters list? http://ardupilot.org/copter/docs/parameters.html

Hwurzburg, 2019-10-10:
every vehicle has different code/behaviour for handling the various failsafes....first find the parameter like  FS_GCS_ENABLE and then search for that params variable name...like " failsafe_gcs" in Copter...then examine the code...

am closing this since its not a Wiki issue...

closed 2019-10-10 by Hwurzburg