ArduPilot/ardupilot_wiki #2064: Import DroneKit wiki to our developer wiki

rmackay9, 2019-10-11:
I think enough time has passed that we can be pretty sure that without our intervention the DroneKit wiki page won't re-appear.  I think we should make it part of our wiki.

Hwurzburg, 2019-10-25:
closing....links updated

auturgy, 2019-10-25:
Still relevant. This relates to bringing inthe full DK wiki, not just updating hyperlinks.
https://discuss.ardupilot.org/t/dronekit-rescue-project/48505

Hwurzburg, 2019-10-25:
We can discuss at next Wiki call....Randy and Peter (the maintainer) and I had a discussion yesterday...my understanding of the issues were: @peterbarker is the maintainer of the python dronekit and stated that he was keeping up with commercial requests....Randy's biggest issue was that it was maintained (@peterbarker said that he was) , and documentation was missing and unlinked, which has been fixed, and that Ardupilot use in companion computers was still viable, which it apparently is...but we can discuss again in the next formal wiki call , in a week and a half...

brunoolivieri, 2019-10-25:
my 2 cents:

1) As a light user of copters with dronekit I do not feel secure with the way that is supported today. I think that would be wonderful to have it internally, as "dronecode is inside its ecosystem".  (I know that dronekit and dronecode are agnostic enough to work with any vehicle that works with mavlink. )

I think that having dronekit formally under Ardupilot tutelage it could bring advantages for actual and possible future partners interest to sell equipments with SBC onboard capabilities.

2) I like this subject very much. I have some SBC around (edison, rpi, bbb and dragonboard 410c), Copters, and some basic python and lua previous trials. At least in the documentation and field tests I could help somehow.

peterbarker, 2019-10-25:
On Fri, 25 Oct 2019, Henry Wurzburg wrote:

> keeping up with commercial requests....Randy's biggest issue was that it was

Clarification: working on things (non-dronekit-python-related) which 
people are sometimes paying me to do.

rmackay9, 2019-10-26:
After a discussion with @peterbarker, I think we don't need to import the dronekit docs unless we takeover dronekit completely.  I think we should only consider taking over dronekit if we are willing to support [this proposal](https://discuss.ardupilot.org/t/dronekit-python-rescue-project/48505) to put some work into it... that's just my view though.