ArduPilot/ardupilot_wiki #2071: Rover, Plane, Copter: add in-flight compass learning explanation

rmackay9, 2019-10-12:
Copter, Plane and Rover-4.0.0 include "inflight learning of compass offsets".  We should add a new section to [this wiki page](http://ardupilot.org/rover/docs/common-compass-calibration-in-mission-planner.html) to clarify to users how it can be used.

I've briefly tested it in SITL on a Rover and it seems to work and then on [this discussion](https://discuss.ardupilot.org/t/rover-4-0-0-rc1-available-for-beta-testing/47706/30) I passed the following advice to a user.

This is the procedure:

- set COMPASS_LEARN = 3
- the message “CompassLearn: Initialised” will appear on the MP’s message tab (it does not appear in red letters on the HUD).
- “Bad Compass” will appear but this is nothing to be worried about. We will hopefully make this disappear before the final release
- Arm and drive the vehicle around in whatever mode you like, do some turns
- “CompassLearn: have earth field” should appear on MP’s message tab and then eventually “CompassLearn: finished”.
- if you want you can check the COMPASS_LEARN parameter has been set back to zero (you may need to refresh parameters to see this) and the COMPASS_OFS_X/Y/Z values will have changed.

Note that this is a simple calibration of offsets only. The 3D compass calibration does a better job because it calculations scaling as well. But for cases where it’s not possible to lift and rotate the vehicle hopefully this new method will be better than nothing.

rmackay9, 2019-10-12:
we should probably put a warning on the page that this new method is only available on versions 4.0.0 (and higher).

rmackay9, 2019-10-12:
@tridge tells me that we've got an RC option available for this feature as well.

Hwurzburg, 2019-10-12:
Questons: COMPASS_LEARN....1 is only  for plane, 2 for any vehicle....saved on disarm...is this a one shot deal, resetting to  COMPASS_LEARN =0 after disarm? or continuous? are all just offsets? I understand that  onboard or live methods will also detect orientation and scaling...correct?
does the RCx_OPTION version just run the inflight and save on exit?

rmackay9, 2019-10-12:
Some answers:

- I'm pretty sure compass-learn=3 doesn't do orientations.  It also doesn't do scaling, it just does offsets.
- once the compass-learn-3 completes it resets the compass-learn parameter to 0.  In my test in SITL this only takes a couple of turns so it can complete within a minute.  I haven't tested on a real vehicle though.
- Copters should never use any of these methods - normal users should stick with the regular "Onboard Calibration" method.  The new method hasn't been proven to work yet on copters in wide variety of situations.  We may invite beta testers to try it and provide logs so we can determine if it's worked.
- Large rovers than can't use the regular Onboard method are invited to use the new compass-learn-3 method.

Hwurzburg, 2019-10-15:
PR #2088 addresses this....

closed 2019-11-01 by Hwurzburg