ArduPilot/ardupilot_wiki #2082: Omnibus F4 Nano V6 (must be V6) voltage monitor

cjmv, 2019-10-13:
[Omnibus F4 Nano V6 (must be V6)](http://ardupilot.org/copter/docs/common-omnibusnanov6.html).

I’m having some trouble getting the Battery monitor to work. As soon as I try to activate it I get “This feature is not enabled in your firmware”.
I'm really not sure if it ever worked before because this is a new build but, I have recollection that before version 3.6.10 (so... maybe 3.6.9) I think I saw it working.

Should I try to re-flash it?
I'm attaching the full param list. in case it helps.

By the way, just a small detail on the diagram at http://ardupilot.org/copter/docs/common-omnibusnanov6.html, I believe the VTX should be connected to the video output and camera to the input and not the other way around.

Thanks in advance for the help,

Carlos Vicente

[param_list.txt](https://github.com/ArduPilot/ardupilot_wiki/files/3721535/param_list.txt)

vierfuffzig, 2019-10-13:
to enable battery monitoring on a freshly flashed board use the full parameter list and set:

BATT_MONITOR = 3 (analog voltage only) and reboot.

you'll then find the pin values default to:

BATT_VOLT_MULT = 11
BATT_VOLT_PIN = 12

your pin values have been set to -1 (= disabled). check above BATT_VOLT parameters and set as required.

updated hardware instructions respectively and fixed image video connections.

thanks for reporting.

cjmv, 2019-10-13:
That did the trick! To be quite honest I don't even remember touching the BATT_VOLT_PIN parameter!
Thanks for the help!

closed 2019-10-13 by cjmv